openwisp/openwisp-monitoring #1: Delete monitoring objects when devices are deleted

nemesisdesign, 2020-03-20:
When a device is deleted, its related metric and check (eg: ping) are not deleted.
Ideally we should delete the related objects automatically and make it easy to recover these via django-reversion.

nepython, 2020-03-21:
I am working on this :smiley:

nepython, 2020-03-22:
@nemesisdesign, will it be fine if I create a custom delete action because I thought it would be best to display the related metrics and check which shall get deleted, so this will help to modify the **delete selected devices** page to display the same.

nemesisdesign, 2020-03-26:
@nepython it should be done with a post_delete signal, hooked in the ready method of this class: https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/check/apps.py

Like here: https://github.com/openwisp/openwisp-controller/blob/master/openwisp_controller/connection/apps.py#L15-L29

PR #9 (2020-04-15): [enhancement] Delete monitoring objects when devices are deleted
Closes #1
closed 2020-04-15 by nemesisdesign