openwisp/openwisp-monitoring #13: [monitoring] Move check into monitoring app

nemesisdesign, 2020-04-04:
We should move check in `openwisp_monitoring.monitoring` since it makes sense that it's part of the main monitoring solution. It's still good to keep the code separated, but it should be configured in a way that it shows up under the monitoring app and not in its own app.

We should make sure any DB migration generated by this change performs the migration in a backward compatible way.

Purhan, 2021-02-16:
@nemesisdesign I'm not sure if my implement is headed in the right direction, please check the draft PR above

nemesisdesign, 2021-03-27:
I think there's no real necessity of doing this. I will close for now, if the need arises we'll reopen.

closed 2021-03-27 by nemesisdesign