openwisp/openwisp-monitoring #26: [monitoring] Graphs: allow specifying range of dates an grouping by time

nemesisdesign, 2020-04-13:
Allow specify by custom date range: the user will be able to select a range of dates and reload the graphs for the dates specified.

First the backend and API must be changed, then the HTML template and JS code.
The grouping of the timeseries must adjust to the length of the range automatically.

The grouping is defined here: https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/monitoring/models.py#L310

For example, right now, 1 day intervals group results by 10 minutes. While 3 day intervals group results by 20 minutes.

The system could determine the optimum grouping automatically, ranging from 5 minutes (specific days) up to 24 hours (ranges up to 1 year).

The minimum range is 1 day, the maximum range is 1 year.

yashikajotwani12, 2022-06-29:
I am using https://www.daterangepicker.com/ for enabling the range of dates.

**Note**: After we add this section of selecting dates, this current section of  `1 day`, `3 days`, and so on, can be removed, and instead we would be able to access this through `today`, `yesterday`, and so on.

[image]

[image]

[image]

[image]

yashikajotwani12, 2022-06-30:
**Update**: 

[image]

https://user-images.githubusercontent.com/77020164/176736217-e6ed088f-d1b9-47ea-8a3b-2393f78c0bbc.mp4

nemesisdesign, 2022-06-30:
@yashikajotwani12 good progress, can you please invert the position of export data and the date widget?

Can you also make sure that these buttons look good on narrow screens too? What happens on mobile phones or tables?

yashikajotwani12, 2022-06-30:
> @yashikajotwani12 good progress, can you please invert the position of export data and the date widget?
> 
> Can you also make sure that these buttons look good on narrow screens too? What happens on mobile phones or tables?

Yeah! I will invert the buttons, and also check for different dimensions :+1:

commit d7f81942 (2022-07-05): [feature] Added dates inside the widget #26

updated the widget, added each date
functionality inside the widget.

Fixes #26
commit 5c4d11fa (2022-07-05): [feature] Charts: allow specifying range of dates #26

Fixes #26
commit e6679087 (2022-07-05): [feature] Added dates inside the widget #26

updated the widget, added each date
functionality inside the widget.

Fixes #26
commit 0eb79355 (2022-07-05): [feature] Updated for the custom datepicker #26

Fixes #26
closed 2022-10-25 by nemesisdesign