openwisp/openwisp-monitoring #27: [monitoring] Zooming graphs: reload data in order to provide a detailed view

nemesisdesign, 2020-04-13:
Depends on https://github.com/openwisp/openwisp-monitoring/issues/26.

Let's saying we're viewing a graph for 1 year of data and we zoom in to 2 specific days in which an incident happened.

Right now we wouldn't see much, because the zoom will simply enlarge the current monthly graph, which is grouping results by 2 hours, so we'll simply see 2 big bars.

The ideal behavior would be to reload the data of those 2 days by leveraging the work that will be done in #26.

yashikajotwani12, 2022-08-03:
**ToDos:**

1. **On zooming one year of data from [12th July to 17th July]** [Before]
[image]

     **Converted into a 7-day dateSpan Graph** [After]
[image]
**IMP** :
     **This Zooming view to show data in contracted form would be temporary on refreshing or resetting the axis we would get the old data-time for the zoomed chart.**

2. **On sliding  left we want the previous data to load as we advance.**

    https://user-images.githubusercontent.com/77020164/182622713-ae6de6c0-d70f-4f4b-99bc-23ef9d8aba03.mp4

3. **We want this  sliding feature to happen with all the charts simultaneously.**

yashikajotwani12, 2022-08-09:
Progress:

https://user-images.githubusercontent.com/77020164/183686412-a098b030-d78e-46e4-b46b-1992ef387423.mp4

nemesisdesign, 2022-08-09:
> Progress:
> 
>  ScreenRecorder_2022-08-09_b1900e18-efb4-4cbd-bfd4-200404a73c3e.mp4

Nice :+1:, keep us posted.

commit d262776a (2022-10-21): [feature] Added zooming graphs to provide a detailed view #27

Closes #27
commit e83cf985 (2022-10-25): [feature] Added zooming graphs to provide a detailed view #27

Closes #27
commit d5b15c6e (2022-10-25): [feature] Added zooming graphs to provide a detailed view #27

Closes #27
PR #450 (2023-01-20): [feature] Added zooming graphs to provide a detailed view #27
-  Added custom zooming to monitoring charts. 
- `Zoom In` : To get a detailed view of all the monitoring charts for a specific time range. 
- `Zoom Out` : To reset all monitoring charts to the last selected time range.

### `Demo`

https://user-images.githubusercontent.com/56113566/198017360-a56d1ba2-fe28-4738-9886-eb0851cc04dc.mp4

Closes #27

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
closed 2023-01-20 by nemesisdesign