openwisp/openwisp-monitoring #30: [monitoring] Allow to store IP addresses of interfaces in DeviceData.data

nemesisdesign, 2020-04-14:
The device status information (accessible via `DeviceData.data`) stores information about the current status of the device (uptime, memory usage, load averages).

When the device has interfaces (eg LAN or WAN) set in DHCP client mode, there's no way to easily know which address the device got from the DHCP server.

This aims to fix that.

Incoming payoad example:
```
{
    'type': 'DeviceMonitoring',
    'interfaces': [
        {
            'name': 'br-lan',
            'addresses': [
                {
                    'proto': 'static',
                    'family': 'ipv4',
                    'address': '192.168.1.1',
                    'mask': 24
                },
                {
                    'proto': 'dhcp',
                    'family': 'ipv4',
                    'address': '10.0.0.4',
                    'mask': 24
                }
            ]
        }
    }
}
```

The schema in `openwisp_monitoring.device.schema` will have to be updated to include the addresses schema: http://netjson.org/rfc.html#addresses1
The gateway option can be ignored.

If we get a non-wireless interface without any address, we will not store it for now (wireless interfaces are stored anyway, they contain other useful info related to the wireless settings).

Once the data is saved and validated correctly (a few tests will need to ensure bogus addresses will be rejected) we will have to ensure the ip addresses of each interface are shown in the device status tab:
https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/device/templates/admin/config/device/change_form.html#L66-L124

nepython, 2020-05-01:
I am working on it :smiley:

nemesisdesign, 2020-05-01:
@nepython do you already have an openwrt device or VM? Coordinate on the gitter channel with @NoumbissiValere and @PabloCastellano, they already have done similar things.

nepython, 2020-05-01:
>@nepython do you already have an openwrt device or VM? 

No, I don't have an openwrt device currently but I have setup a VM (though I need to configure it correctly).
Is it essential for this PR, if so then I will proceed with configuring it.
> Coordinate on the gitter channel with @NoumbissiValere and @PabloCastellano, they already have done similar things.

Sure :smile:

nepython, 2020-05-01:
>When the device has interfaces (eg LAN or WAN) set in DHCP client mode, there's no way to easily know which address the device got from the DHCP server.

@nemesisdesign, can you please give few details on how the `addresses` need to be refactored to easily retrieve them later. Should I do something similar to what has been done for `uptime`.
https://github.com/openwisp/openwisp-monitoring/blob/4d96d743e0e583ba10445b08d2ed34b18d1be45a/openwisp_monitoring/device/models.py#L55-L57
I have created a draft PR [here](https://github.com/openwisp/openwisp-monitoring/pull/64). I am still working on it, created it for reference.

nemesisdesign, 2020-05-01:
@nepython just loop over the addresses and show them in a table, eg:

``
address          |   mask      |   DHCP client?  
192.168.1.2    |  /{mask}   |   yes / no
```

The difficult part may be to get the correct `mask`, but we'll deal with that on the openwrt side.

PR #64 (2020-05-14): [monitoring] Allow to store IP addresses of interfaces in DeviceData.data #30
~~This is still WIP. The following need to be done:-~~
* [x] show the addresses in the device status tab: https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/device/templates/admin/config/device/change_form.html
* [x] add tests to ensure bogus addresses will be rejected
* [x] Creating a PR in `monitoring-lua` as mentioned in the issue

**PS**: The JSON type for `family` was provided as `integer` [here](http://netjson.org/rfc.html#addresses1) but it appeared visibly as `string` to me

Close [... cut]
closed 2020-05-14 by nemesisdesign