openwisp/openwisp-monitoring #34: [monitoring] Allow collecting ARP table information in DeviceData.data

nemesisdesign, 2020-04-14:
Like https://github.com/openwisp/openwisp-monitoring/issues/33, but with ARP table information.

Incoming ARP table data:

```python
{
    'type': 'DeviceMonitoring',
    'arp_table': [
        {
            'ip_address': '192.168.66.197',  # ipv4 or ipv6
            'mac_address': 'f2:f1:3e:56:d2:66',  # mac address
            'interface': 'br-lan',  # string
            'type': '0x1',  # string
            'flags': '0x2',  # string
            'mask': '*',  # string
        }
    }
}
```

The schema in `openwisp_monitoring.device.schema` will have to be updated  accordingly.

Once the data is saved and validated correctly (a few tests will need to ensure bogus arp info will be rejected) we will have to ensure the arp table info is shown in the device status tab:
https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/device/templates/admin/config/device/change_form.html#L66-L124

If the arp table info is not available, do not mention this in the admin UI.

PabloCastellano, 2020-04-30:
@nemesisdesign I'm close to finishing ARP table data collection. What kind of tests are appropriate to include? As I understand, JSON-schema validates the data received from the device but the validation is very simple (e.g. it does not check if the string received in the `ip_address` field is an IP address indeed).

nemesisdesign, 2020-04-30:
@PabloCastellano it should be possible to specify that it should be an ip, don't worry about it now, let's do a few iterations

PR #62 (2020-05-08): [feature] Allow collecting ARP table information in DeviceData.data #34
[image]

Closes #34
commit 809914be (2020-05-08): [feature] Allow collecting ARP table information in DeviceData.data #34

* Enable json schema format checker

Closes #34
closed 2020-05-08 by nemesisdesign