openwisp/openwisp-monitoring #46: [enhancement] Renaming the models Threshold and Graph in a backward compatitable manner

nepython, 2020-04-21:
Rename the models which are now called `Threshold`, `Graph` to `AlertSettings`, `Chart` respectively, in a backward compatible way (eg: provide a data migration which creates the alert table first, copies the data from the threshold table into alert and then remove the old threshold table, or an equivalently working alternative).

PabloCastellano, 2020-05-25:
Just as a side note for the backwards-compatible part, Django migrations support renaming models/tables.

nepython, 2020-05-25:
@PabloCastellano, I forgot to share a mailing list thread. I noted the same [here](https://groups.google.com/d/msg/openwisp/tinsyzJt3BE/3VazszdaAwAJ)

PabloCastellano, 2020-05-25:
Since Django 2.0 was released on 2nd December 2017 and the first commit in this repo came 2 months later, I think it's safe to assume that everyone using this module is using Django 2.0+ no there's no need to take extra caution

nemesisdesign, 2020-05-29:
> Since Django 2.0 was released on 2nd December 2017 and the first commit in this repo came 2 months later, I think it's safe to assume that everyone using this module is using Django 2.0+ no there's no need to take extra caution

:+1:

commit 8293fc37 (2020-06-12): Rename models Threshold, Graph to AlertThreshold, Chart #46

Closes #46
commit 31cb73de (2020-06-12): [chores] Renamed Graph to Chart and Threshold to AlertSettings #46

Closes #46
commit 0c053636 (2020-06-12): [chores] Renamed Graph to Chart and Threshold to AlertSettings #46

Closes #46
PR #127 (2020-06-12): Rename models Threshold, Graph to AlertThreshold, Chart #46
Now testing this in staging.

Closes #46
closed 2020-06-12 by nemesisdesign