openwisp/openwisp-monitoring #52: [feature] Add an Alert Settings section in device admin

nepython, 2020-04-27:
Add an `Alert Settings` section in the device admin which allows to see what alerts are available for a `device` and which also allow to modify the `threshold` or **disable/enable** the alarm.

nepython, 2020-06-27:
I was trying to figure out what can be done here. It won't be as simple as #53 (PR #148), as we need to display alertsettings related to a device but there doesn't exist any direct relation between these two models. They are related by metric and django doesn't provide an easy way to do this.

There are few options that we may use:
* We may add three fields to the alertsettings that map to the monitoring model's fields namely `content_type`, `object_id`, `content_object`. Then by using GenericTabularInline we achieve what we want (in my opinion adding these fields seems a bit redundant, not sure.)
* We create our own Inline that shall take into account all three models Metric, AlertSettings, Device and display accordingly. These seems a good approach but we will end up creating our own `js, css, view, admin, formsets, tests` ( too much work but what we will be doing is just repeating the code because it is not fully extensible. )
* The third approach is `NestedInline`, this is a [feature request almost 12 years old](https://code.djangoproject.com/ticket/9025), there are many third party apps which provide this feature. I tried out [django-nested-admin](https://github.com/theatlantic/django-nested-admin) since it's been maintained well and is there from around 2013.
[image]

The first and third approach (adds a dependency) seem better to me, in case there is some other better way please let me know, I have tried almost everything I could get. Inlines have many limitations too like `list_display` doesn't work other wise I could have tried adding a column. 
Please let me know which approach seems better?

PabloCastellano, 2020-06-27:
Since AlertSettings.metric already links to the Metric model and Metric contains a GenericForeignKey. Did you try defining a [reverse generic relation](https://docs.djangoproject.com/en/3.0/ref/contrib/contenttypes/#reverse-generic-relations)?

From the documentation:

> The relation on the related object back to this object doesn’t exist by default. Setting **related_query_name** creates a relation from the related object back to this one. This allows querying and filtering from the related object.

This way you can track back [ Device ]-- Generic FK---->[ Metric ]----- FK ----->[ AlertSettings ]

nepython, 2020-06-28:
Reverse generic relations need to be declared in order to use `GenericTabularInline`, so that's not a problem here. I am not sure if I get you correctly but `related_query_name` is to be used for reverse querying as explained in https://stackoverflow.com/a/36166644.

This is the structure currently, I tried tracing it back but the only way it worked was by declaring the three fields as mentioned in https://github.com/openwisp/openwisp-monitoring/issues/52#issuecomment-650563847
```python
class AbstractMetric(TimeStampedEditableModel):
    content_type = models.ForeignKey(
        ContentType, on_delete=models.CASCADE, null=True, blank=True
    )
    object_id = models.CharField(max_length=36, db_index=True, blank=True)
    content_object = GenericForeignKey('content_type', 'object_id')
    # ...

class AbstractAlertSettings(TimeStampedEditableModel):
    metric = models.OneToOneField(
        get_model_name('monitoring', 'Metric'), on_delete=models.CASCADE
    )
    # ...

class DeviceData(AbstractDeviceData, Device):
    checks = GenericRelation(get_model_name('check', 'Check'))
    metrics = GenericRelation(get_model_name('monitoring', 'Metric'))
    # ...
```
Let me know if I am missing anything.

PabloCastellano, 2020-06-28:
@nepython What is the current problem you are trying to solve? By reading your comment I think you are mixing multiple ones.

Let's go step by step. Can you answer the following questions?

1. Are you able to retrieve the metrics associated to a device?
2. Are you able to retrieve the Alertsettings for those metrics?
3. Did you find any issue displaying the alertsettings fields in the device page?

Also notice that the generic FK from Metric points to the Device model, not DeviceData.

https://github.com/openwisp/openwisp-monitoring/blob/23f3105c539950d1f547f5dcefcda1228c09bcf1/openwisp_monitoring/device/api/views.py#L191-L200

nepython, 2020-06-28:
> 1. Are you able to retrieve the metrics associated to a device?

Yes
> 2. Are you able to retrieve the Alertsettings for those metrics?

Not entirely, alertsettings can be pointed to as a readonly field only. If I put it as a editable field then it simply raises an error, no such field (I find it weird as to why is this the case).
> 3. Did you find any issue displaying the alertsettings fields in the device page?

I am not able to retrieve alertsettings fields via the inline.
There are many limitations, few are:
* alertsettings__value won't work
* `list_display` which we can normally use in a ModelAdmin to add custom columns (eg. health_status is device change_list doesn't work in inline, etc.)

> Also notice that the generic FK from Metric points to the Device model, not DeviceData. Is that the issue?

Thanks for correcting me but that's not the issue.

PabloCastellano, 2020-06-28:
@nepython Ok. I think I understand you better now. Now, can you provide a draft Pull Request showing some AlertSettings fields if possible? So I can help you iterating.

PR #150 (2020-07-25): [feature] Add an Alert Settings section in device admin #52
### Blocked by #162
Closes #52, #67
This is how it looks, right now:
[image]
commit 4f226192 (2020-07-25): [feature] Added "Alert Settings" section in device admin #52

Implements and closes #52
commit 40465c2e (2020-08-07): [feature] Added "Alert Settings" section in device admin #52

Implements and closes #52
closed 2020-08-07 by nemesisdesign