openwisp/openwisp-monitoring #55: [feature] Add a Check which inspects links and nodes & creates relevant charts for the same

nepython, 2020-04-27:
The issue can be broken down as:
### 1. Add check, metric and a relative alert settings
Add a `check`, `metric` and a `relative alert settings` that will be available only if `openwisp-network-topology` is in **INSTALLED_APPS** (created automatically for every topology, it must take into account existing **topologies** by using a data migration) that is run periodically (by default every 5 minutes).
### 2. The Check should:-
* write to the timeseries DB the number of links that are down, the number of links that are up, the total number of nodes and the number of isolated nodes (nodes without links).
* check if the topology has **X %** or more of its links in **down** state, in which case it should send an alert. **X** should be configurable with a django setting defaulting to **50%**.
### 3. Add Relevant charts
Two charts should be created: one which shows the amount of links (one line for the **up** state and one line for the **down** state) and another chart for the amount of nodes (one line for total nodes and one line for isolated nodes).
### 4. Documentation
Document this feature and add illustrative pictures of the `charts`.

nepython, 2020-07-02:
I was able to create a topology and update it's strategy thus creating relevant links and nodes.
I have few **queries** below as I don't have a deep understanding of how this works :sweat_smile:.

* How to link a device to a `Topology`? (I mean by passing `last_ip` as `gateway id` or anything similar so later on we can find topology related to a device).

> Two charts should be created: one which shows the amount of links (one line for the up state and one line for the down state) and another chart for the amount of nodes (one line for total nodes and one line for isolated nodes).

* I think the former chart should be similar to `uptime` chart while the latter might be similar to `rtt` chart (not sure). I don't know what is meant by `isolated nodes`, a brief description please.

I have these queries due to lack of deeper understanding of networks, I am looking for relevant resources to understand key concepts but am unable to classify what is useful and what is not. So, if you have any resources to suggest, please do it. Thanks! :smile:

nemesisdesign, 2020-07-04:
@nepython let's put this in standby. Do not work further on this please.

nemesisdesign, 2020-07-25:
I'm closing this. Not needed right now. We'll work on other things.

closed 2020-07-25 by nemesisdesign