openwisp/openwisp-monitoring #60: [monitoring] Show associated wifi clients

nemesisdesign, 2020-04-29:
We already receive the information of the connected wifi clients, which is used to generate the graph which shows how many wifi clients connect over time.

It would be very useful to also display the list of clients for each interface.

We already show some status information about the wifi interface, eg: `INTERFACE STATUS: WLAN0`. We could add a new section within this table which shows the list of clients, each row could have columns like ma address, wds, ht, vht, wmm, wps.

NoumbissiValere, 2020-04-30:
@nemesisdesign I created a wireless interface on my `OpenWRT VM` from my `openwisp-monitoring` instance by updating the `template` we've been using. I expected the `Device status` tab to be updated with the `interface information` which i will need to update. But i saw none after the `cron` job's elapse time. Please am I missing something?

nemesisdesign, 2020-04-30:
@NoumbissiValere we need to understand why you're not getting any info for the interfaces.
If you look at the lua script, you'll find there's some code which tries to get info from the interfaces.

Try executing the following commands on OpenWRT and let me know what output you get:

```
ubus call network.device status
ubus call network.wireless status
ubus call iwinfo info '{"device": "<wireless_interface_name_here>"}'
ubus call hostapd.<wireless_interface_name_here> get_clients
```

substitute `<wireless_interface_name_here>` with the wireless interface name.

NoumbissiValere, 2020-05-07:
@nemesisdesign `ubus call network.wireless status` have been giving me an empty dict as output. and consequently, `ubus call iwinfo info ...` and `ubus call hostapd ...` has been telling me `command failed`. 
I checked the `openwrt` device on the browser and it shows the wireless interface i created for the template has been successfully applied to the device.  still can't figure out what am missing. here is the interface i added to the template

```"interfaces": [
        {
            "wireless": {
                "network": [],
                "mode": "access_point",
                "radio": "radio0",
                "ack_distance": 0,
                "rts_threshold": 0,
                "frag_threshold": 0,
                "ssid": "HelloWorld",
                "hidden": false,
                "wds": true,
                "encryption": {
                    "protocol": "none"
                },
                "wmm": true,
                "isolate": false,
                "macfilter": "disable",
                "maclist": []
            },
            "type": "wireless",
            "name": "wlan0",
            "mtu": 1500,
            "mac": "30-65-EC-6F-C4-58",
            "autostart": true,
            "disabled": false,
            "addresses": [],
            "network": ""
        }
    ],
    "radios": [
        {
            "protocol": "802.11n",
            "name": "radio0",
            "phy": "",
            "channel": 0,
            "channel_width": 20,
            "tx_power": 0,
            "country": "00",
            "disabled": false,
            "driver": "mac80211",
            "hwmode": "11g"
        }
    ]```

nemesisdesign, 2020-05-08:
@NoumbissiValere is the package `rpcd-mod-iwinfo` installed (reboot after installing, install with `opkg update` and `opkg install rpcd-mod-iwinfo`)? Is it a device or is it the VM?

Anyway, if you don't manage to get it working, you can simulate it by sending this data in POST to the device metric view:

```json
{
    "type": "DeviceMonitoring",
    "general": {
        "hostname": "Test-01",
        "local_time": 1588910240,
        "uptime": 226021
    },
    "interfaces": [
        {
            "name": "wlan0",
            "statistics": {
                "collisions": 0,
                "multicast": 0,
                "rx_bytes": 1421451,
                "rx_compressed": 0,
                "rx_crc_errors": 0,
                "rx_dropped": 0,
                "rx_errors": 0,
                "rx_fifo_errors": 0,
                "rx_frame_errors": 0,
                "rx_length_errors": 0,
                "rx_missed_errors": 0,
                "rx_over_errors": 0,
                "rx_packets": 10062,
                "tx_aborted_errors": 0,
                "tx_bytes": 9681874,
                "tx_carrier_errors": 0,
                "tx_compressed": 0,
                "tx_dropped": 0,
                "tx_errors": 0,
                "tx_fifo_errors": 0,
                "tx_heartbeat_errors": 0,
                "tx_packets": 120242,
                "tx_window_errors": 0
            },
            "wireless": {
                "channel": 36,
                "country": "US",
                "frequency": 5180,
                "mode": "access_point",
                "noise": -103,
                "signal": -56,
                "ssid": "wifi-test",
                "tx_power": 23,
                "clients": [
                    {
                        "aid": 1,
                        "assoc": true,
                        "auth": true,
                        "authorized": true,
                        "ht": true,
                        "mac": "30:d4:56:19:3b:49",
                        "mfp": false,
                        "preauth": false,
                        "rrm": [
                            0,
                            0,
                            0,
                            0,
                            0
                        ],
                        "signature": "wifi4|probe:0,1,45,127,191,221(0050f2,8),255,127,255,221(506f9a,22),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:3391f9fa,vhtrxmcs:030cfffa,vhttxmcs:030cfffa,extcap:00000a82004000000001|assoc:0,1,45,59,127,191,221(0050f2,2),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:3380f1b2,vhtrxmcs:030cfffa,vhttxmcs:030cfffa,extcap:0400000200000040",
                        "vht": true,
                        "wds": false,
                        "wmm": true,
                        "wps": false
                    }
                ]
            }
        }
    ],
    "resources": {
        "load": [
            0.03,
            0.01,
            0
        ],
        "memory": {
            "available": 55361536,
            "buffered": 4042752,
            "cached": 11923456,
            "free": 71524352,
            "shared": 167936,
            "total": 127500288
        },
        "swap": {
            "free": 0,
            "total": 0
        }
    }
}
```

NoumbissiValere, 2020-05-08:
@nemesisdesign i am using a VM and not a device. the packages were already installed. i used the POST data and got results. I have send a PR. Please can you review it.

PR #74 (2020-05-15): [monitoring] Show associated wifi clients #60
Fixed the issue by updating change_form template

Fixes #60
commit 792a0322 (2020-05-15): [monitoring] Show associated wifi clients #60 (#74)

Fixes #60
closed 2020-05-15 by nemesisdesign