openwisp/openwisp-monitoring #65: [enhancement] Abstract code communicating with the time series DB

nepython, 2020-05-05:
Abstract the code that communicates with the timeseries DB so that new timeseries backends can be written to support other timeseries DBs.

nepython, 2020-05-27:
I am thinking of separating all the queries and calling the same from a dictionary. The dictionary will contain the queries for both the timeseries which shall make it easier to use across the module. This will organize all the queries in one **dict** object but it will also result in the code of many objects getting scattered (example, **DEFAULT_CHARTS**). This might not be a good way to do it but I can't think of any other way. So is this correct or there is some better way to do it?

PabloCastellano, 2020-05-27:
@nepython The approach looks good to me

nepython, 2020-05-28:
> @nepython The approach looks good to me

@PabloCastellano, #103 should give you a better view of how I was thinking of implementing it.
Please have a look :smile:

PR #103 (2020-06-23): [enhancement] Abstract code communicating with the time series DB #65
* I have added few **TODO** comments as reminder notes when adding support for another timeseries db
* Updated the docs on how to select / declare a timeseries database in `settings.py`
* Introduced Deprecation warning to shift to the latest format of defining the timeseries db

Closes #65
commit b2553bae (2020-06-23): [db] Abstracted code which deals with time series DB #65

Implements and closes #65
closed 2020-06-23 by nemesisdesign