openwisp/openwisp-monitoring #66: [enhancement] Add support for another timeseries DB

nepython, 2020-05-05:
Add support for another timeseries DB, it could be `prometheus`, `opentsdb` or another good option if there's any.
The decision regarding which one to choose must be backed up by convincing research.

Check the discussion [here](https://groups.google.com/forum/#!topic/openwisp/tinsyzJt3BE)

Points to remember:
- we need to implement the current features and make them both work for `Prometheus` and `Influxdb`
- the new timeseries DB that we will support needs to provide an **open source solution for horizontal scaling**, because influxdb does not provide it (has a paid cloud service instead)

**NOTE**: This requires #65 to be complete

nepython, 2020-05-22:
Found this https://github.com/prometheus/influxdb_exporter. It might make things easier. :smile:

PabloCastellano, 2020-05-23:
We should discuss how to properly setup the metrics exporter, as I have different ideas but I don't know what is the best at the moment.

1. Creating a exporter to be installed in the device is the standard, however due to device limited resources sometimes it might not be the best idea
2. Using something like the pushgateway

There are already exporters for cpu, load, memory, network, etc. They are usually written in Golang, resulting in a quite big executable to put on small devices. So we would have to implement our own ones... in Lua? Doesn't sound very exciting reinventing the wheel in Lua.

By using pushgateway we can reuse the Lua code we have already written.

These are my views, I will try to get some time to think about it so that we can properly discuss it.

nemesisdesign, 2020-07-10:
Closing, let's continue the discussion in #99.

closed 2020-07-10 by nemesisdesign