openwisp/openwisp-monitoring #68: [enhancement] Prepare travis-ci build

nepython, 2020-05-05:
Prepare a `travis-ci` build which runs tests with both timeseries DB supported, performs QA checks (like other equivalent openwisp modules) and collects test coverage data.

The issue can be broken down too:
* [x] Prepare a basic travis-ci build that collects test coverage data(It has already been done thanks to @pandafy)
* [x] Travis also stands updated with qa checks and includes black formatting (thanks to @PabloCastellano  and @nemesisdesign)
* [ ] Prepare travis to run tests for both Timeseries DB

To be done after #66

nemesisdesign, 2020-08-31:
@nepython what's the status of this?

nepython, 2020-08-31:
> @nepython what's the status of this?

It has been mostly completed in other PRs, the last part of adding support for elasticsearch so we can run tests with both the Timeseries databases together has been implemented in #164 (still open but this part of Travis is completed in it).

closed 2021-04-06 by nemesisdesign