openwisp/openwisp-monitoring #69: Optimize unit test runtime speed

nepython, 2020-05-05:
Locate slow tests (tests taking longer than 0.3 secs) and find out if mocking can be used to optimize them.
These can be helpful pointers:
* Checkout [locate slow tests](https://github.com/ninuxorg/nodeshot/blob/master/nodeshot/core/base/tests_speedup.py), it should help to locate slower tests.
* Check `--parallel` provided provided by django for testing (provided that it does not affect test-coverage sharply )
* and mocking whatever possible (but not duplicating the entire logic :sweat_smile: )

Just for the record it took 
```C
Ran 142 tests in 37.713s
```
on my latest travis-build

nepython, 2020-06-14:
### Test Speed (before mocking [travis-build](https://travis-ci.org/github/openwisp/openwisp-monitoring/jobs/697732991#L1037))
* sample-app : 38.726s
* actual apps : 45.147s

### Test Speed (after mocking [travis-build](https://travis-ci.com/github/nepython/openwisp-monitoring/jobs/348881143#L1068))
* sample-app : 21.090s
* actual apps : 22.457s

**Note**: Figures vary by 2-3 second on individual builds but this is the general trend.

**Query**: What is the ideal test speed to be achieved. I mean I know it's best to keep all the the  tests below a threshold (0.3s is optimal target, I think). I tried best to achieve this goal but due to the number of API requests that few tests need to make, the test slows down (I think `get/post` requests for those tests (test_api, test_admin) shouldn't be mocked).
So, please let me know the ideal target. Suggestions to speed up these requests.

**Suggestion**: I think we can include [TimeLoggingRunner](https://github.com/nepython/openwisp-monitoring/pull/6/files#diff-09d5bb4c3393cecf48149f97c79cc9a4) in openwisp-utils to measure test speeds for various tests. The reason being that when introducing a new test, we ordinarily don't measure how fast it is (eg, I introduced `test_trigger_device_recovery_task` which took earlier approx. **5s** on travis but it went unnoticed until now). Finding slow tests by modifying test runner is quite easy as we change only one setting in Django.

nemesisdesign, 2020-06-15:
> **Note**: Figures vary by 2-3 second on individual builds but this is the general trend.
>
> **Query**: What is the ideal test speed to be achieved. I mean I know it's best to keep all the the tests [...]

We don't have hard and set rules. The faster the better as long as we don't sacrifice catching bugs by mocking too much.

Staying below 0.3 is great.

Correct, requests to our own application should not be mocked.

Only network interactions with external systems.

I would focus on locating the slow tests

> **Suggestion**: I think we can include [TimeLoggingRunner](https://github.com/nepython/openwisp-monitoring/pull/6/files#diff-09d5bb4c3393cecf48149f97c79cc9a4) in openwisp-utils to measure test [...]

Yes that's great, please go ahead and add it to `openwisp_utils.tests` and document it in the README.

PR #134 (2020-06-21): [tests] Optimize unit test runtime speed #69
Closes #69
closed 2020-06-21 by nemesisdesign