openwisp/openwisp-monitoring #70: [docs] Update README

nepython, 2020-05-05:
Ensure the `README` is up to date, lists all the features available, provides some `screenshots` and a detailed explaination of the usage of the main features added in this project.

nepython, 2020-05-05:
I will surely document as I introduce the changes but this issue is for updating the **README** to make it more effective for the end-user to understand the module :smile:

nemesisdesign, 2020-05-19:
Let's coordinate with @atb00ker since he's been doing this for different modules and we've been experimenting with animated gifs which seem a great way to show what the module can do at glance.

@pandafy related to you as well :point_up:

atb00ker, 2020-05-19:
My workflow for creating the GIFs, 
1. List all the features in order that I want to show them in the video.
2. Do some demo runs to ensure that GIF is as short as possible in the final shot.
3. Use Firefox, because F11 takes you to full screenwithout any borders.
4. Use `simplescreenrecorder` recording `mkv` in full screen at 7 FPS
5. Convert mkv to an efficient GIF: 
```
ffmpeg -i  vid.mkv -vf fps=7,scale=1366:-1:flags=lanczos,palettegen palette.png
ffmpeg -i vid.mkv -i palette.png -filter_complex "fps=7,scale=1366:-1:flags=lanczos[x];[x][1:v]paletteuse" demo.gif
```

6. Use https://ezgif.com/optimize to optimize further according to requirement.
7. If the size is still big, some steps specific to the GIF may be required.

atb00ker, 2020-05-19:
@nepython @pandafy , let me know if you need any further assistance with creating GIFs! :smile:

nepython, 2020-05-19:
Thanks, @atb00ker :smiley:

nemesisdesign, 2020-05-19:
That's great, thanks a lot @atb00ker :clap: :clap:

nemesisdesign, 2020-07-24:
**Update**: in the next weeks we should revise the README to ensure it's complete and all screenshots are up to date, especially the animated gif.

PR #209 (2020-08-28): [docs] Update Readme #70

Checks:

- ~~[ ] I have manually tested the proposed changes~~
- ~~[ ] I have written new test cases to avoid regressions (if necessary)~~
- [x] I have updated the documentation (e.g. README.rst)

* [x] Badges as other OpenWISP modules
* [x] `run_checks` command (maybe something similar to https://github.com/openwisp/django-freeradius/blob/master/docs/source/general/management_commands.rst)
* [x] Signals Missing:
 * pre_metric_write
 * post_metric_write
 * health_status_changed
* [ ] Images (`AlertSettings Inline`, `Checks Inline`, `Metric Health`) and GIF need to be  [... cut]
commit bf71f80d (2020-08-28): [docs] Updated README #70

Closes #70
closed 2020-08-31 by nemesisdesign