openwisp/openwisp-monitoring #80: QuotaExceededError when setting items in localStorage

PabloCastellano, 2020-05-18:
The following line is throwing error `QuotaExceededError: The quota has been exceeded` when I'm browsing the device graphs

https://github.com/openwisp/openwisp-monitoring/blob/196bac1467daaa85af1adb1e5ef3de12619528b9/openwisp_monitoring/device/templates/admin/config/device/change_form.html#L309

nemesisdesign, 2020-05-18:
@PabloCastellano I'm surpirsed, it should be a simple key with a very short value.
Is this happening because for some bug the timeRangeKey is always changing and the application is storing many different keys? Or is it because the browser localStorage is full and any additional key will trigger the error?

nemesisdesign, 2022-04-04:
I am closing because we could not replicate this issue, we'll open again if it reappears.

closed 2022-04-04 by nemesisdesign