openwisp/openwisp-monitoring #86: [charts] Add way to register and unregister new chart configurations

nemesisdesign, 2020-05-19:
Just like we're doing in https://github.com/openwisp/openwisp-notifications/pull/19/files#diff-e4c93b8a95c5a88dcf0c7b41e5c5ff5fR1

nepython, 2020-06-05:
### Up for discussion
@nemesisdesign @PabloCastellano
After this feature, `OPENWISP_MONITORING_CHARTS` seems a bit redundant to me, hence I suggest that we remove it.
In case, I am missing anything, please correct me :)

PabloCastellano, 2020-06-08:
@nepython I agree on removing ADDITIONAL_CHARTS but I think we can keep OPENWISP_MONITORING_CHARTS so that we don't need to call get_chart_configuration and check that the fields are correct every time.

nepython, 2020-06-09:
>@nepython I agree on removing ADDITIONAL_CHARTS but I think we can keep OPENWISP_MONITORING_CHARTS so that we don't need to call get_chart_configuration and check that the fields are correct every time. WDYT?

I think, I didn't understand it fully. Please let me know if the current approach seems fine :smiley:
**+1** to not removing `OPENWISP_MONITORING_CHARTS`. I think it is still useful, use case: if the user wants to modify just one field in existing chart configuration, it seems more better approach.

PR #117 (2020-07-07): [charts] Add way to register and unregister new chart configurations
Closes #86
commit 6da80beb (2020-07-07): [charts] Add way to register and unregister new chart configurations #86

Closes #86
commit ebc41fe1 (2020-07-15): [charts] Add way to register and unregister new chart configurations #86

Closes #86
commit 3e929169 (2020-07-17): [charts] Add way to register and unregister new chart configurations #86

Closes #86
closed 2020-08-07 by nemesisdesign