openwisp/openwisp-monitoring #87: [enhancement] Use template tags to make it more readable

nepython, 2020-05-20:
Use the template tag [filesizeformat](https://docs.djangoproject.com/en/3.0/ref/templates/builtins/#filesizeformat) for `Monitoring Graphs`.
It formats the value to a `human-readable` file size (i.e. **13 KB**, **4.1 MB**, **102 bytes**, etc.).

For example:
```js
{{ value|filesizeformat }}
If value is 123456789, the output would be 117.7 MB.
```
There's also another template tag [humanize](https://docs.djangoproject.com/en/3.0/ref/contrib/humanize/) which helps to make it more human-readable.

We need to flag a chart in some way such that it's bytes are formatted automatically.
Currently, the values of the chart have been formatted as per the units.
This works well mostly, but sometimes it doesn't. For example, when not much traffic is being consumed, the points of the graph show numbers like **0.004 GB** which looks very bad.

nemesisdesign, 2020-07-01:
@nepython the problem is that this transformation has to be done also for the values displayed in the charts

nemesisdesign, 2021-03-27:
Here's the requirements for this issue:

- we want to have the traffic graphs show either B, KB, MB, GB, TB depending on the size of the numbers
- not only in the chart summary (total traffic consumed for the selected period) but also in each data point
- it should not be hardcoded into the traffic charts but we should be able to flag a chart to have its value pass through a filter, like a `conversion` flag in the chart config, which could have multiple values, each value representing a filter, but initially we only implement this filter 

Example traffic chart config:

```python
{
    'traffic': {
        'label': _('Traffic'),
        'name': '{name}',
        'key': '{key}',
        'field_name': 'rx_bytes',
        'related_fields': ['tx_bytes'],
        'charts': {
            'traffic': {
                'type': 'scatter',
                'title': _('Traffic: {metric.key}'),
                'label': _('Traffic'),
                'description': _(
                    'Network traffic, download and upload, measured on '
                    'the interface "{metric.key}"'
                ),
                'summary_labels': [
                    _('Total download traffic'),
                    _('Total upload traffic'),
                ],
                'order': 240,
                'query': chart_query['traffic'],
                # added this, removed "unit"
                'conversion': {
                    'filter': 'bytes',
                },
            }
        },
    },
}
```

Adding `conversion` as a dictionary allows to define arguments that can be passed to the filter if needed.

atb00ker, 2022-06-25:
No data scenario seems to be working fine! 😄 

[image]

commit 948aad35 (2022-06-19): [ux] Show size in (KB, MB or GB) adaptively in charts #87

Minor Improvements

Fixes #87
commit ce0a6131 (2022-06-19): [ux] Show size in (KB, MB or GB) adaptively in charts #87

Fixes #87
commit 7330c416 (2022-06-20): [ux] Show size in (KB, MB or GB) adaptively in charts #87

Improved percircles unit
Fixes #87
commit 0fc15790 (2022-06-22): [ux] Show size in (KB, MB or GB) adaptively in charts #87

Fixes #87
closed 2022-07-19 by nemesisdesign