openwisp/openwisp-monitoring #89: [qa] Use Github pull templates

PabloCastellano, 2020-05-20:
Using pull templates like [explained here](https://help.github.com/en/github/building-a-strong-community/about-issue-and-pull-request-templates) we can improve our workflow and assure more quality in our PRs by default.

The template should contain at least the following checkboxes:

- [ ] I have updated the documentation (e.g. README.rst)
- [ ] I have written new test cases to avoid regressions

nemesisdesign, 2020-05-20:
@PabloCastellano what if there's no need to update the docs? Should we have a case for that or just contemplate it with the same line like " I have updated the documentation in README.rst or no documentation update is needed"?

PabloCastellano, 2020-05-20:
I don't see any problem in leaving the checkbox unmarked if there's no documentation to update. At least it makes it clearer that documentation is something to be updated unless necessary

nemesisdesign, 2020-05-20:
Maybe we could add a line like "I have read and I am following the [OpenWISP contributing guidelines](http://openwisp.io/docs/developer/contributing.html#how-to-commit-your-changes-properly)"?

PabloCastellano, 2020-05-20:
Maybe we can recommend reading it, but I would not make it a checkbox. I don't even remember myself all the guidelines :-D

nemesisdesign, 2020-05-20:
Fair enough. As long as the link is there, some people will read it.
At some point we will also simplify and shorten the contribution guidelines.

commit 4672db74 (2020-07-07): feat(qa): use GitHub Pull Request templates #89

Closes #89
commit 4851035c (2020-07-07): feat(qa): use GitHub Pull Request templates #89

Closes #89
PR #159 (2020-07-08): feat(qa): use GitHub Pull Request templates #89
Let me know what you think about the text. We can improve it now or later

Closes #89
commit cb28a13b (2020-07-08): [qa] Use GitHub Pull Request templates #89

Closes #89
closed 2020-07-08 by nemesisdesign