openwisp/openwisp-monitoring #100: [prometheus] Metrics exporter for openwrt

PabloCastellano, 2020-05-26:
We can't use Golang's Prometheus node exporter because the binaries are too big for the openwrt devices. Fortunately @nepython has found https://stuarthowlette.me.uk/posts/prometheus-monitoring-other-os/, a node exporter written in lua with many plugins so that you can add more metrics. Looks like the way to go but before we arrive to this task, let's take the time to analyze all possible choices.

* [ ] Decide what exporter to use (or even implement our own if necessary)
* [ ] Analyze if the metrics exposed by the existing exporters cover our current needs
* [ ] Value how easy is to contribute back to the existing projects in case we need to implement our own metrics
* [ ] Document how to install the exporter in an openwrt device

2stacks, 2020-05-26:
Is fluentbit an option?

PabloCastellano, 2020-05-26:
@2stacks From fluentbit homepage:

> Fluent Bit is a Fast and Lightweight Log Processor and Forwarder for Linux, OSX and BSD family operating systems.

As I understand, Fluent bit would be helpful to stream logs but not metrics. Besides that, I have compiled fluent-bit myself and the binary weights 6MB, which is also a lot for most of openwrt devices. So I would say it's not an option at the moment, but let me know if you think I'm wrong

nepython, 2020-05-26:
I checked out the docs/mailing-list and noted the below points.
* There happen to be to be two tools fluentd and fluent-bit. While Fluentd consumes around 40 MB, Fluent Bit was designed for high performance and comes with a super light footprint, running on ~450KB only (specially made for IoT devices is a newer project but comes with almost 10 times fewer plugins support)
* Fluent-bit runs well on openwrt. Concluded from this [thread](https://groups.google.com/forum/#!msg/fluent-bit/u9scILPoPr0/d33h7JUlBQAJ)
* Both come with native support for Prometheus. [Using fluentd with prometheus](https://docs.fluentd.org/deployment/monitoring-prometheus).

Finally, I think that fluentbit is a nice tool which can be used for many things but in this case the package I have suggested is very specific for our purpose (consumes around ~10kb as informed by @PabloCastellano) and thus might be a lighter dependency to add.

@2stacks I haven't used fluent-bits. Hence, concluded this based upon what I have read, please correct me in case I miss anything and thanks for the suggestion :smile:

PabloCastellano, 2020-05-26:
@nepython You are confusing metrics with logs. Check out https://www.sumologic.com/blog/logs-metrics-overview/

nemesisdesign, 2020-07-25:
As discussed recently, we decided to proceed with Elasticsearch (see #99), hence I'm closing this issue to clean up the issue list. Thanks!

closed 2020-07-25 by nemesisdesign