openwisp/openwisp-monitoring #101: [metrics] Measurement stored in InfluxDB aren't deleted when related metrics are deleted.

nepython, 2020-05-26:
When a metric is deleted the related measurement in the influxdb is not deleted, ideally that should be the case. Since, it might lead to lesser data stored in the timeseries db
**Possible Solution**: Using a pre-delete signal and sending relevant query to InfluxDB

**Up for Discussion**
* Should we delete it or not?
* If not we shall document this behavior

PabloCastellano, 2020-05-26:
Documentation: https://docs.influxdata.com/influxdb/v1.8/guides/downsample_and_retain/

PabloCastellano, 2020-06-07:
I tend to think that we should not delete old data. Maybe we can add a message making it clear that deleting a Metric object will not delete its stored data in the database.

IMHO the quantity of data we already store is insignificant and we shouldn't worry at this point about making such optimizations.

But event though I have said that, this could be an improvement that we could do in the future. The user could choose to delete the stored data too when deleting a Metric object

nemesifier, 2023-10-03:
Should have been solved by https://github.com/openwisp/openwisp-monitoring/pull/429.

closed 2023-10-03 by nemesifier