openwisp/openwisp-monitoring #106: [alert settings] Add way to specify global defaults

nemesisdesign, 2020-05-29:
This is a draft and is open for discussion.

Right now, threshold objects are created whenever a metric is created.
If we want to change its default values globally (value, operator, seconds), we would need to loop over all the available thresholds and change each one.

I've been thinking, what if we allow these fields to be NULL and we define their default values in the metric configuration (#44)?
For example, metrics that generate an alert, will specify this in their configuration, and will specify the default values of the fields that can be NULL.
We can then add model methods that retrieve the value, something like:

```python
def get_value(self):
    # if self.value is None, the default from the configuration will be used
    return self.value or self.config['value']
```

A solution like the one I described here would give us the possibility to change the defaults globally without having to change the database.

The user can then override specific thresholds by setting their own custom value if needed (eg: a device is particularly problematic and the user will want to change some parameters).

I also think we may use the chance to rename `seconds` to something more obvious, maybe `tolerance`.

We should take advantage to document how thresholds/alert settings work.

Related to #44 and #52.

nepython, 2020-08-02:
> **Example**: the metrics being added in #76, we have different threshold values for memory, CPU, disk. What if a user having already 100 deployed devices wants to set the threshold value of the memory to 75%? With the change I'm proposing here the user would be able to do it just by changing a setting.
> 
> I've been thinking, what if we allow these fields to be NULL and we define their default values in the metric configuration (#44)?
> For example, metrics that generate an alert, will specify this in their configuration, and will specify the default values of the fields that can be NULL.
> We can then add model methods that retrieve the value, something like:
> 
> ```python
> def get_value(self):
>     # if self.value is None, the default from the configuration will be used
>     return self.value or self.config['value']
> ```
> 
> A solution like the one I described here would give us the possibility to change the defaults globally without having to change the database.

**QUERY**
@nemesisdesign, wouldn't the user still need to clear the `values` column in order for this to work. I mean, let us assume the user has 100 devices with 100 corresponding `memory metrics` and `alert settings`. Now, if the user wants to change the `values` of all these 100 alert settings then won't he first need to clear the existing `values` column and set all of them to `NULL`? Please correct me in case, I am misunderstanding anything here.
> The user can then override specific thresholds by setting their own custom value if needed (eg: a device is particularly problematic and the user will want to change some parameters).
> ### Find a better name for the `seconds` field
> 
> I also think we may use the chance to rename `seconds` to something more obvious, maybe `tolerance`.

I am not sure but the `tolerance` might also be confused for `value` (to be renamed to `threshold`). How about `threshold_time` or `threshold_seconds` ? They are not technical terms but might be more easier to understand, I think.
> ### AlertSettings.threshold
> 
> Now that Threshold has been renamed to `AlertSettings`, it makes sense to rename `AlertSettings.value` to `AlertSettings.threshold`.

:+1:
> ### Notification types
> 
> The goal of this section is to allow us to have more user friendly alerts/notifications.

I am not fully sure of how it's being done currently but will check out.
> ### Documentation
> 
> We should take advantage to document how thresholds/alert settings work.
> ### Conclusion
> 
> I think we should look for a way to improve harmonize the way metrics, charts, threshold and notifications are defined and extended (eg: the register/unregister actions) because in openwisp-monitoring they're closely related.
> 
> We can take the chance of making everything clearer, more consistent, more documented and easier for the user to understand and use.
> 
> **This will be a great improvement to this module!**.
> ### Notes during our call
> 
> http://www.catb.org/esr/writings/taoup/html/ch09s01.html
> 
> Example usage:
> 
> ```python
> register_thing([
>     {
>         "name": "ping",
>         # ...
>         # ...
>         "charts": [
>             {"name": "uptime"},
>             {"name": "packet loss"},
>             {"name": "rtt"}
>         ],
>         "alert": {
>             "operator": "<",
>             "threshold": 1,
>             "tolerance": 500
>         },
>         "notification": {
>             ""
>         }
>     }
> ])
> ```
:+1:

nepython, 2020-08-03:
@nemesisdesign, I was on a draft for this and I think these issues,
* After merging `chart config` in `metric config`, I think `chart_config`, all associated functions and settings can also be safely migrated to work with the new well defined `metric_config` in order to avoid duplication and simplify the structure (wanted to confirm this, before moving forward.)
> ```python
>def get_value(self):
>    # if self.value is None, the default from the configuration will be used
>    return self.value or self.config['value']
>```

* This works great but what about the AlertSettings admin? I mean it will show `blank` values for `null` fields (which will look bad :sweat_smile:). A good solution might be overriding `full_clean` and storing `config` values as default (similar to what we do for `key`, `field_name` in Metric). Should I go ahead with this?

nepython, 2020-08-03:
@nemesisdesign, here is the [draft for the ping metric configuration](https://gist.github.com/nepython/100f93f6fe7c0b639dfcffdeca1a1d4b) I was thinking of for this issue. Please suggest improvements / corrections. Thanks! :smile:

nemesisdesign, 2020-08-04:
> > **Example**: the metrics being added in #76, we have different threshold values for memory, CPU, disk. What if a user having already 100 deployed devices wants to set the threshold value of the memory to 75%? With the change I'm proposing here the user would be able to do it just by changing a setting.
> > I've been thinking, what if we allow these fields to be NULL and we define their default values in the metric configuration (#44)?
> > For example, metrics that generate an alert, will specify this in their configuration, and will specify the default values of the fields that can be NULL.
> > We can then add model methods that retrieve the value, something like:
> > ```python
> > def get_value(self):
> >     # if self.value is None, the default from the configuration will be used
> >     return self.value or self.config['value']
> > ```
> > 
> > 
> > A solution like the one I described here would give us the possibility to change the defaults globally without having to change the database.
> 
> **QUERY**
> @nemesisdesign, wouldn't the user still need to clear the `values` column in order for this to work. I mean, let us assume the user has 100 devices with 100 corresponding `memory metrics` and `alert settings`. Now, if the user wants to change the `values` of all these 100 alert settings then won't he first need to clear the existing `values` column and set all of them to `NULL`? Please correct me in case, I am misunderstanding anything here.
> **PS**: I am referring this case for existing metrics and alert settings (clarifying to avoid confusions )

We have to create a data migration which compares the existing alert settings to the global default settings. If the value in the DB is equal to the default value configured, we can set it to NULL.

> I am not sure but the `tolerance` might also be confused for `value` (to be renamed to `threshold`). How about `threshold_time` or `threshold_seconds` ? They are not technical terms but might be [...]

I think we can do the following:

- rename value to threshold and use "threshold value" as its label and provide a good help_text
- rename seconds to tolerance and use "threshold tolerance" as its label and provide a good help_text, we may also change it to be based in minutes rather than seconds, to simplify the user experience
 
> > ### Notification types
> > The goal of this section is to allow us to have more user friendly alerts/notifications.
> 
> I am not fully sure of how it's being done currently but will check out.

The example you provided seems to be on track.
 
[...]

> @nemesisdesign, I was working on a draft for this and wanted to get a confirmation on the following before moving forward,
> 
> * After merging `chart config` in `metric config`, I think `chart_config`, all associated functions and settings can also be safely migrated to work with the new well defined `metric_config` in order to avoid duplication and simplify the structure (wanted to confirm this, before moving forward.)

I have not understood this very well, can you provide some more context and or an example?
However, as well as the change you're thinking of makes things better in some way (eg: more consistent, reduced duplication) you're on the right track.

> * This works great but what about the AlertSettings admin? I mean it will show `blank` values for `null` fields (which will look bad sweat_smile). A good solution might be overriding `full_clean` [...]

If we store something other than NULL, changing the default values from the global configuration without having to run data migrations won't be possible.

I would "fake" the UI in some way so that if the value is NULL, the defaults are shown. This shouldn't be hard, it should be possible to do it with a custom django model form or a similar solution.

On the model side instead, we can tweak the validation so that if the default values are passed, internally they will be converted to `None`.

> @nemesisdesign, here is the [draft for the ping metric configuration](https://gist.github.com/nepython/100f93f6fe7c0b639dfcffdeca1a1d4b) I was thinking of for this issue. Please suggest improvements / corrections. Thanks!

Looks good, remember to make all strings that are meant to be shown in the UI translatable.
`alert` should become `alert_settings`, `threshold_crossed` and `threshold_recovery` may be shortened to `problem` and `recovery`.

pandafy, 2020-08-04:
> @nemesisdesign, here is the [draft for the ping metric configuration](https://gist.github.com/nepython/100f93f6fe7c0b639dfcffdeca1a1d4b) I was thinking of for this issue. Please suggest improvements / corrections. Thanks!

@nepython you can only have one notification type with a name, i.e. there can not be multiple occurrence of 'threshold_crossed' or 'threshold_recovery'. **openwisp-notifications** will raise exception while registering these notification types for multiple metrics. 

@nemesisdesign defining these many notification types, might make the whole point of having notification types pointless. I think it was aimed to bring generalization in creating notifications. Please let me know if I am wrong, If not, we might have to rethink notification types(notification message) feature.

nepython, 2020-08-04:
I was testing the newly added notification types (as per the issue description) and the output already appears very nice thanks to the work of @pandafy on `openwisp-notifications`. I am currently facing three issues,
<p align="center">
  [image]
</p>

* Can we configure the format in which timestamp is being displayed?
* Timestamp is not taking care of my timezone (timestamp is currently displayed in `<UTC>`). @nemesisdesign, I also noted that in the issue description, you have specifically put `<UTC datetime>`, any reason for it? Since, I think the user might be interested in getting time reference of his timezone, wdyt?
* The threshold value for resources metric is appearing as a `float`, I tried to simply use a f-string to make it appear as a percent but it didn't work :/
Below, is the message
```python
'message': _(
    'The device "[{notification.target}]" {notification.verb} disk usage which has gone '
    'over {notification.actor.alertsettings.threshold}, '  # tried using {notification.actor.alertsettings.threshold*100}%
    'since {notification.timestamp}.'
),
```
A simple solution might be to store all percentages as a multiple of 100 (so a threshold of `0.80` --> `80`).
* Also, I can't recall where but we had a discussion to rename resource metrics `field_name` to have a consistency in this issue, so should I proceed with it?

nemesisdesign, 2020-08-04:
@pandafy 

> @nepython you can only have one notification type with a name, i.e. there can not be multiple occurrence of 'threshold_crossed' or 'threshold_recovery'. **openwisp-notifications** will raise [...]

The recurring keys will have to be named differently internally, we need a new key called `type_name` for example.
 
> @nemesisdesign defining these many notification types, might make the whole point of having notification types pointless. I think it was aimed to bring generalization in creating notifications. [...]

I don't understand why do you think it's pointless?
The point of creating notification types is to be able to categorize them so we can easily refer to them and manipulate their properties (eg: description, email subject and soon also default notification settings).
I do not understand what you mean for "generalization", but for sure we didn't do all this work to keep these incredibly ugly notification messages. We need to make the notification messages more user friendly and tailored for every situation.
"Threshold of object X has crossed" has to go, it has to be replaced with something like "Device X is not reachable anymore", "Memory usage on deevice X is over 95%" and so on.

nemesisdesign, 2020-08-04:
@nepython 

> I was testing the newly added notification types (as per the issue description) and the output already appears very nice thanks to the work of @pandafy on `openwisp-notifications`. I am currently facing three issues,
> 
> [image]
> 
> * Can we configure the format in which timestamp is being displayed?
> * Timestamp is not taking care of my timezone (timestamp is currently displayed in `<UTC>`). @nemesisdesign, I also noted that in the issue description, you have specifically put `<UTC datetime>`, any reason for it? Since, I think the user might be interested in getting time reference of his timezone, wdyt?

There's surely a way but let's remove it for now since it looks redundant.

> * The threshold value for resources metric is appearing as a `float`, I tried to simply use a f-string to make it appear as a percent but it didn't work :/
>   Below, is the message

Look at how it's implemented: https://github.com/openwisp/openwisp-notifications/blob/master/openwisp_notifications/base/models.py#L39-L91
It's using `string.format()` so you can't use f-strings, you should be able to do it by passing the format recognized by `string.format()`, see:

- https://docs.python.org/3.4/library/string.html#formatspec
- https://pyformat.info 
 
> ```python
> 'message': _(
>     'The device "[{notification.target}]" {notification.verb} disk usage which has gone '
>     'over {notification.actor.alertsettings.threshold}, '  # tried using [...]

Yes but prepare a separate PR for that, do not mix it here.

pandafy, 2020-08-04:
@nemesisdesign 

I agree that we should provide user friendly notification message. Maybe, I got the impression of notification types wrong. :sweat_smile: 

> The recurring keys will have to be named differently internally, we need a new key called `type_name` for example.

I have opened https://github.com/openwisp/openwisp-notifications/issues/88 for this, please let me know if I got requirements right. :grinning:

nemesisdesign, 2020-08-04:
> @nemesisdesign
> 
> I agree that we should provide user friendly notification message. Maybe, I got the impression of notification types wrong. 
> 
> > The recurring keys will have to be named differently internally, we need a new key called `type_name` for example.
> 
> I have opened [openwisp/openwisp-notifications#88](https://github.com/openwisp/openwisp-notifications/issues/88) for this, please let me know if I got requirements right.

We'll have to handle that internally here when looping over the configuration and calling the register notification type function.

PR #185 (2020-08-18): [enhancement] Add way to specify global defaults

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [x] I have updated the documentation (e.g. README.rst)

Closes #106
commit 791b7cd7 (2020-08-18): [feature] Add way to specify global defaults #106

Implements and closes #106
commit c53d80a0 (2020-08-28): [feature] Add way to specify global defaults #106

Implements and closes #106
closed 2020-08-31 by nemesisdesign