openwisp/openwisp-monitoring #108: [urgent] Checks stopped working

nemesisdesign, 2020-05-29:
Active checks (ping) stopped working after #102 has been merged.

This is the output from celery:

```
[2020-05-29 05:05:20,728: INFO/MainProcess] Received task: openwisp_monitoring.check.tasks.run_checks[a616fd68-0593-4271-ad74-4b2e6544273d]  
[2020-05-29 05:05:20,729: DEBUG/MainProcess] TaskPool: Apply <function _fast_trace_task at 0x7f694c1e69e0> (args:('openwisp_monitoring.check.tasks.run_checks', 'a616fd68-0593-4271-ad74-4b2e6544273d', {'lang': 'py', 'task': 'openwisp_monitoring.check.tasks.run_checks', 'id': 'a616fd68-0593-4271-ad74-4b2e6544273d', 'shadow': None, 'eta': None, 'expires': None, 'group': None, 'retries': 0, 'timelimit': [None, None], 'root_id': 'a616fd68-0593-4271-ad74-4b2e6544273d', 'parent_id': None, 'argsrepr': '()', 'kwargsrepr': '{}', 'origin': 'gen3164@envy', 'reply_to': '76a72368-5591-3387-be9e-cb2d26bd5386', 'correlation_id': 'a616fd68-0593-4271-ad74-4b2e6544273d', 'hostname': 'celery@envy', 'delivery_info': {'exchange': '', 'routing_key': 'celery', 'priority': 0, 'redelivered': None}, 'args': [], 'kwargs': {}}, b'[[], {}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]', 'application/json', 'utf-8') kwargs:{})
[2020-05-29 05:05:20,730: DEBUG/MainProcess] Task accepted: openwisp_monitoring.check.tasks.run_checks[a616fd68-0593-4271-ad74-4b2e6544273d] pid:3159
[2020-05-29 05:05:20,739: INFO/ForkPoolWorker-6] Task openwisp_monitoring.check.tasks.run_checks[a616fd68-0593-4271-ad74-4b2e6544273d] succeeded in 0.009382204996654764s: None
[2020-05-29 05:07:57,472: ERROR/MainProcess] Received unregistered task of type 'openwisp_monitoring.device.tasks.trigger_device_checks'.
The message has been ignored and discarded.

Did you remember to import the module containing this task?
Or maybe you're using relative imports?

Please see
http://docs.celeryq.org/en/latest/internals/protocol.html
for more information.

The full contents of the message body was:
b'[[], {"pk": "cc5c1d61-8e0b-4f31-9ab4-70cf6bb0c55b"}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]' (121b)
Traceback (most recent call last):
  File "/home/nemesis/.virtualenvs/openwisp2/lib/python3.7/site-packages/celery/worker/consumer/consumer.py", line 562, in on_task_received
    strategy = strategies[type_]
KeyError: 'openwisp_monitoring.device.tasks.trigger_device_checks'

```

Notice how `trigger_device_checks`fails.

If I reset the repository back to commit ID a427878, everything works:

```
[2020-05-29 05:10:32,600: INFO/MainProcess] Received task: openwisp_monitoring.check.tasks.perform_check[e951d1ab-9529-4d5f-b2f1-7a00a2272247]  
[2020-05-29 05:10:32,601: DEBUG/MainProcess] TaskPool: Apply <function _fast_trace_task at 0x7f7d5f28d9e0> (args:('openwisp_monitoring.check.tasks.perform_check', 'e951d1ab-9529-4d5f-b2f1-7a00a2272247', {'lang': 'py', 'task': 'openwisp_monitoring.check.tasks.perform_check', 'id': 'e951d1ab-9529-4d5f-b2f1-7a00a2272247', 'shadow': None, 'eta': None, 'expires': None, 'group': None, 'retries': 0, 'timelimit': [None, None], 'root_id': '336ed862-cae6-4904-b34d-baf9f2e89e21', 'parent_id': '336ed862-cae6-4904-b34d-baf9f2e89e21', 'argsrepr': "(UUID('a02bca26-3c63-4794-919b-e26c6cc14df4'),)", 'kwargsrepr': '{}', 'origin': 'gen3591@envy', 'reply_to': '33a194ba-3517-3bce-b124-044f29765dfc', 'correlation_id': 'e951d1ab-9529-4d5f-b2f1-7a00a2272247', 'hostname': 'celery@envy', 'delivery_info': {'exchange': '', 'routing_key': 'celery', 'priority': 0, 'redelivered': None}, 'args': ['a02bca26-3c63-4794-919b-e26c6cc14df4'], 'kwargs': {}}, b'[["a02bca26-3c63-4794-919b-e26c6cc14df4"], {}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]', 'application/json', 'utf-8') kwargs:{})
[2020-05-29 05:10:32,602: DEBUG/MainProcess] Task accepted: openwisp_monitoring.check.tasks.perform_check[e951d1ab-9529-4d5f-b2f1-7a00a2272247] pid:3592
[2020-05-29 05:10:36,774: WARNING/ForkPoolWorker-7] Content-Type: text/plain; charset="utf-8"
MIME-Version: 1.0
Content-Transfer-Encoding: 7bit
Subject: Metric "Ping (device: ca
From: webmaster@localhost
To: operator@test.com
Date: Fri, 29 May 2020 03:10:36 -0000
Message-ID: <159072183677.3592.4699459685698384389@envy>

Metric "Ping (device: cap-dev-01)" returned within threshold limit.
[2020-05-29 05:10:36,775: WARNING/ForkPoolWorker-7] -------------------------------------------------------------------------------
[2020-05-29 05:10:36,782: WARNING/ForkPoolWorker-7] {
    "loss": 0.0,
    "reachable": 1,
    "rtt_avg": 8.49,
    "rtt_max": 14.2,
    "rtt_min": 2.05
}
[2020-05-29 05:10:36,782: INFO/ForkPoolWorker-7] Task openwisp_monitoring.check.tasks.perform_check[e951d1ab-9529-4d5f-b2f1-7a00a2272247] succeeded in 4.180682772974251s: None
[2020-05-29 05:12:24,635: INFO/MainProcess] Received task: openwisp_monitoring.device.tasks.trigger_device_checks[ba685ed7-6982-4b2f-958e-84bf619c767b]  
[2020-05-29 05:12:24,635: DEBUG/MainProcess] TaskPool: Apply <function _fast_trace_task at 0x7f7d5f28d9e0> (args:('openwisp_monitoring.device.tasks.trigger_device_checks', 'ba685ed7-6982-4b2f-958e-84bf619c767b', {'lang': 'py', 'task': 'openwisp_monitoring.device.tasks.trigger_device_checks', 'id': 'ba685ed7-6982-4b2f-958e-84bf619c767b', 'shadow': None, 'eta': None, 'expires': None, 'group': None, 'retries': 0, 'timelimit': [None, None], 'root_id': 'ba685ed7-6982-4b2f-958e-84bf619c767b', 'parent_id': None, 'argsrepr': '()', 'kwargsrepr': "{'pk': UUID('cc5c1d61-8e0b-4f31-9ab4-70cf6bb0c55b')}", 'origin': 'gen3555@envy', 'reply_to': '46164025-b740-3f8e-8a36-92446f3d68ea', 'correlation_id': 'ba685ed7-6982-4b2f-958e-84bf619c767b', 'hostname': 'celery@envy', 'delivery_info': {'exchange': '', 'routing_key': 'celery', 'priority': 0, 'redelivered': None}, 'args': [], 'kwargs': {'pk': 'cc5c1d61-8e0b-4f31-9ab4-70cf6bb0c55b'}}, b'[[], {"pk": "cc5c1d61-8e0b-4f31-9ab4-70cf6bb0c55b"}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]', 'application/json', 'utf-8') kwargs:{})
[2020-05-29 05:12:24,636: DEBUG/MainProcess] Task accepted: openwisp_monitoring.device.tasks.trigger_device_checks[ba685ed7-6982-4b2f-958e-84bf619c767b] pid:3591
[2020-05-29 05:12:24,644: INFO/MainProcess] Received task: openwisp_monitoring.check.tasks.perform_check[976826c7-e0fd-4112-9f14-af072dde10e3]  
[2020-05-29 05:12:24,644: DEBUG/MainProcess] TaskPool: Apply <function _fast_trace_task at 0x7f7d5f28d9e0> (args:('openwisp_monitoring.check.tasks.perform_check', '976826c7-e0fd-4112-9f14-af072dde10e3', {'lang': 'py', 'task': 'openwisp_monitoring.check.tasks.perform_check', 'id': '976826c7-e0fd-4112-9f14-af072dde10e3', 'shadow': None, 'eta': None, 'expires': None, 'group': None, 'retries': 0, 'timelimit': [None, None], 'root_id': 'ba685ed7-6982-4b2f-958e-84bf619c767b', 'parent_id': 'ba685ed7-6982-4b2f-958e-84bf619c767b', 'argsrepr': "(UUID('a02bca26-3c63-4794-919b-e26c6cc14df4'),)", 'kwargsrepr': '{}', 'origin': 'gen3591@envy', 'reply_to': '33a194ba-3517-3bce-b124-044f29765dfc', 'correlation_id': '976826c7-e0fd-4112-9f14-af072dde10e3', 'hostname': 'celery@envy', 'delivery_info': {'exchange': '', 'routing_key': 'celery', 'priority': 0, 'redelivered': None}, 'args': ['a02bca26-3c63-4794-919b-e26c6cc14df4'], 'kwargs': {}}, b'[["a02bca26-3c63-4794-919b-e26c6cc14df4"], {}, {"callbacks": null, "errbacks": null, "chain": null, "chord": null}]', 'application/json', 'utf-8') kwargs:{})
[2020-05-29 05:12:24,645: DEBUG/MainProcess] Task accepted: openwisp_monitoring.check.tasks.perform_check[976826c7-e0fd-4112-9f14-af072dde10e3] pid:3592
[2020-05-29 05:12:24,658: INFO/ForkPoolWorker-6] Task openwisp_monitoring.device.tasks.trigger_device_checks[ba685ed7-6982-4b2f-958e-84bf619c767b] succeeded in 0.022359518974553794s: None
[2020-05-29 05:12:28,725: WARNING/ForkPoolWorker-7] {
    "loss": 0.0,
    "reachable": 1,
    "rtt_avg": 15.5,
    "rtt_max": 35.2,
    "rtt_min": 2.9
}
[2020-05-29 05:12:28,726: INFO/ForkPoolWorker-7] Task openwisp_monitoring.check.tasks.perform_check[976826c7-e0fd-4112-9f14-af072dde10e3] succeeded in 4.08066068199696s: Non
```

nepython, 2020-05-29:
@nemesisdesign, `celery worker` output on my terminal on branch master with latest commit c4defa4
```shell
[2020-05-29 20:00:08,406: WARNING/ForkPoolWorker-6] Content-Type: text/plain; charset="utf-8"
MIME-Version: 1.0
Content-Transfer-Encoding: 7bit
Subject: Metric "Ping (device: 08
From: webmaster@localhost
To: test@test.org
Date: Fri, 29 May 2020 14:30:08 -0000
Message-ID: <159076260840.13642.689655096285074880@Inspiron-5580>

Metric "Ping (device: 08-00-27-04-8e-1c)" returned within threshold limit.
[2020-05-29 20:00:08,406: WARNING/ForkPoolWorker-6] -------------------------------------------------------------------------------
[2020-05-29 20:00:08,429: WARNING/ForkPoolWorker-6] {
    "loss": 0.0,
    "reachable": 1,
    "rtt_avg": 0.45,
    "rtt_max": 0.57,
    "rtt_min": 0.17
}
[2020-05-29 20:00:08,430: INFO/ForkPoolWorker-6] Task openwisp_monitoring.check.tasks.perform_check[1a88a587-794d-4b30-ae5a-5c54b61f4f48] succeeded in 4.400630991000071s: None
[2020-05-29 20:02:00,110: INFO/MainProcess] Received task: openwisp_monitoring.check.tasks.run_checks[ab78613b-43bc-41ac-9a59-c9aee02d3371]  
```
It works perfectly as before I think. @PabloCastellano, can you please check this out since I am unable to reproduce the issue.

nepython, 2020-05-29:
@nemesisdesign, I am unsure but are you running the SAMPLE_APP? Since, I just ran it and noted the following:-
* Celery auto discovers only 4 tasks on the SAMPLE_APP. On the original module, the worker discovers 5 tasks as follows
```
[tasks]
  . openwisp_controller.connection.tasks.update_config
  . openwisp_monitoring.check.tasks.auto_create_ping
  . openwisp_monitoring.check.tasks.perform_check
  . openwisp_monitoring.check.tasks.run_checks
  . openwisp_monitoring.device.tasks.trigger_device_checks
```
The last task (the same one which is giving rise to `KeyError` in your case) isn't discovered.
* CSS and JS aren't getting loaded (especially for the monitoring graphs).
* Migration similar to this https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/check/migrations/0003_create_ping.py is missing in `sample_check`. I had read the using django with celery documentation but didn't encounter it mentioning the need to create a migration file. So, can you please explain why it had been created?

I won't be able to move ahead without some inputs from you since it's working fine on my side besides the `SAMPLE_APP` :sweat_smile:

PabloCastellano, 2020-05-29:
I can't reproduce this issue.

I have pulled from master, reinstalled all dependencies and ran the celery commands and the django server for > 10 minutes

nemesisdesign, 2020-05-30:
> @nemesisdesign, I am unsure but are you running the SAMPLE_APP? Since, I just ran it and noted the following:-

Ok, I confirm that it's happening on the SAMPLE_APP.

> * Celery auto discovers only 4 tasks on the SAMPLE_APP. On the original module, the worker discovers 5 tasks as follows
>
> ```
> [tasks]
>   . openwisp_controller.connection.tasks.update_config
>   . [...]

Ok, this needs to be fixed.

> * CSS and JS aren't getting loaded (especially for the monitoring graphs).

This as well.

> * Migration similar to this https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/check/migrations/0003_create_ping.py is missing in `sample_check`. I had read the using [...]

It's there to create ping check objects for existing devices.

### Conclusion

The purpose of the sample app is to test that we work we did to add swappable models and extensible classes works, and now it doesn't, **so we have not reached this goal yet**.

The final result must be that everything works in the sample app as well, ok?

nepython, 2020-05-30:
> Ok, I confirm that it's happening on the SAMPLE_APP.

In that case, I did the following and everything works fine.
* Deleted existing `Device`.
* `export SAMPLE_APP=1`
* In the openwrt instance, once again input `shared_secret`
* `./tests/manage.py run_checks`
* Waited for a short while and the device with the `Ping` gets created automatically as it was supposed to.

Some of the steps may be redundant but I noted what I did :sweat_smile:. Also, I don't think this is a bug. Should we mention it in docs? (I don't know, since it will only take place if the developer has previously used the module and using the very same old db.)
> CSS and JS aren't getting loaded (especially for the monitoring graphs).

Unsure why, but after the above steps everything got loaded just fine.

> The purpose of the sample app is to test that we work we did to add swappable models and extensible classes works, and now it doesn't, so we have not reached this goal yet.

I don't know, improvements are needed that's for sure and shall be done with #104 and yes I need to figure out why `openwisp_monitoring.device.tasks.trigger_device_checks` is not being auto-detected by celery. It had been introduced for #14 so besides that it shouldn't break any other functionality :smile:

nemesisdesign, 2020-06-01:
@nepython ok, I will test again as you indicate when trying #104.

commit 911e01c6 (2020-06-05): [swappable] Further improvements #104 #108

Closes #104
Closes #108