openwisp/openwisp-monitoring #111: [influxdb] Review measurements

PabloCastellano, 2020-06-01:
This is more a discussion than a bug.

When querying measurements in influxdb, I can see some general names that apply to all devices (such as load, memory and disk) and other that may change between devices (eth0, eth0_1, eth0_2, eth0_3, wlan0, wlan0_1...) which looks weird to me. Should the interface names be inside a 'network' measurement?

```
$ docker-compose exec influxdb influx -database openwisp2 -execute "SHOW MEASUREMENTS"
name: measurements
name
----
br_lan
device_data
disk
eth0
eth0_1
eth0_2
eth0_3
load
memory
ping
wlan0
wlan0_1
```

nemesisdesign, 2020-06-08:
@PabloCastellano `load` has been renamed to `cpu` in #76 (it's one of the last thing I asked @nepython). Regarding renaming the interface names, I'm not sure. It could be useful, but I would wait until we have a use case that pushes us to do it.

nemesisdesign, 2022-04-23:
To be continued in #363.

closed 2022-04-23 by nemesisdesign