openwisp/openwisp-monitoring #120: [monitoring/checks] Add iperf3 check

nemesisdesign, 2020-06-08:
The goal of this issue is to add a bandwidth test using `iperf3`, using the active check mechanism of openwisp monitoring.

The use case is to perform periodic bandwidth test to measure the max bandwidth available (TCP test) and jitter (UDP).

On a macro level, the check would work this way:

1. openwisp connects to the device (only 1 check per device at time) via SSH and launches iperf3 as a client, first in TCP mode, then in UDP mode, the server is configured via setting, using the -j flag to obtain json output
2. the collected data is parsed and stored as a metric (bandwidth information and jitter)
3. SSH connection is closed

The steps to do are more or less the following:

- create iperf check class, the check must use the connection module of openwisp-controller to connect to devices using SSH
- If a device has no active Connection the check will be skipped and a warning logged
- this check should be optional and disabled by default
- we can run it by default every 12 hours
- allow configuring the iperf server globally and by organization with a setting, eg:

```
OPENWISP_MONITORING_IPERF_SERVERS = {
    '': '<DEFAULT_IPERF_SERVER_HERE>',
    '<org-pk>': <ORG_IPERF_SERVER>'
}
```
- we have to implement a lock to allow only 1 iperf check per server at time: https://docs.celeryproject.org/en/latest/tutorials/task-cookbook.html#ensuring-a-task-is-only-executed-one-at-a-timehttps://stackoverflow.com/questions/12003221/celery-task-schedule-ensuring-a-task-is-only-executed-one-at-a-time/12003293
  that is: for every server available, only 1 check can be performed at a time, so the lock has to take this account when calculating the cache-key
- ssh into device, launch iperf TCP client, repeat for UDP, collect data of both tests in a data structure
- handle failures, if server is down, we can store 0, which would trigger an alert (threshold)
- get or create metric
- get or create chart
- get or create threshold
- save data (tcp max bandwidth, UDP jitter)
- document this check

Saurabh-Mokashi, 2022-03-13:
Hi @nemesisdesign  My name is Saurabh Mokashi, a CSE Undergrad from NITK, India. I am interested to contribute to this project, so kindly guide me through the materials and resources required.

rjmcmahon, 2022-08-11:
I'd suggest using the low duty cycle bounceback test found in i[perf 2](https://sourceforge.net/projects/iperf2/).

nemesisdesign, 2022-08-11:
@rjmcmahon thanks for the suggestion but we're using iperf3, is anything equivalent implemented in iperf3 too?

rjmcmahon, 2022-08-11:
Iperf 3 is poorly named. It's designed to measure networks that support large data sets, CERN to DoE sites. Those testing WiFi would benefit by using iperf 2. TXOPs are a major limiting factor to WiFi responsiveness so the tool used should test more than link capacity.

[Comparison chart](https://iperf2.sourceforge.io/IperfCompare.html)

rjmcmahon, 2022-08-11:
parsing is fairly easy to do with python regexp.  There are examples in the [flows] (https://sourceforge.net/p/iperf2/code/ci/master/tree/flows/flows.py) directory. Flows is also a nice wrapper layer that can be used to feed statistics & plotting. One can convert dictionaries to JSON if needed using standard python json libraries.

`self.regex_traffic = re.compile(r'\[\s+\d+] (?P<timestamp>.*) sec\s+(?P<bytes>\d+) Bytes\s+(?P<throughput>\d+) bits/sec\s+(?P<writes>\d+)/(?P<errwrites>\d+)\s+(?P<retry>\d+)\s+(?P<cwnd>\d+)K/(?P<rtt>\d+) us')
`

Aryamanz29, 2022-10-20:
Solved and Closed by : https://github.com/openwisp/openwisp-monitoring/pull/447

closed 2022-10-20 by Aryamanz29