openwisp/openwisp-monitoring #131: [bug] All logs are being duplicated

nepython, 2020-06-15:
On the master branch of openwisp-monitoring, I found that logs are duplicated as can be seen below. I am observing this for the first time.

[image]

PabloCastellano, 2020-06-15:
This change might be related https://github.com/openwisp/openwisp-monitoring/commit/a51467c1effc3f66ae63431ad2d498c485343129

Since we have a root logger now, we probably have to stop propagating from the other loggers:

```
    'loggers': {
        'django': {
            'handlers': ['console'],
            'level': os.getenv('DJANGO_LOG_LEVEL', 'INFO'),
            'propagate': False,
        },
```

nepython, 2020-06-20:
Closed by #137

closed 2020-06-20 by nepython