openwisp/openwisp-monitoring #156: [metric migration] Chart not found

PabloCastellano, 2020-07-06:
#142 got merged and I had [this issue](https://github.com/openwisp/openwisp-monitoring/pull/142#pullrequestreview-442385967) with the migration. I'm still stuck with my old metrics and I can't migrate to the latest db schema. I can probably delete all data and start again

Trying to migrate **dev** (Chart does not exist):

```
$ ./manage.py migrate
Operations to perform:
  Apply all migrations: account, admin, auth, check, config, connection, contenttypes, device_monitoring, geo, monitoring, notifications, openwisp_notifications, openwisp_users, pki, reversion, sessions, sites, socialaccount
Running migrations:
  Applying monitoring.0013_metric_configuration... OK
  Applying monitoring.0014_merge_traffic_metrics_and_fill_metric_config...Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/base.py", line 369, in execute
    output = self.handle(*args, **options)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/commands/migrate.py", line 233, in handle
    fake_initial=fake_initial,
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/migrations/executor.py", line 245, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/migrations/operations/special.py", line 190, in database_forwards
    self.code(from_state.apps, schema_editor)
  File "/home/pablo/src/openwisp-monitoring/openwisp_monitoring/monitoring/migrations/0014_merge_traffic_metrics_and_fill_metric_config.py", line 24, in merge_traffic_metrics
    chart = Chart.objects.get(metric=tx_metric)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/query.py", line 417, in get
    self.model._meta.object_name
openwisp_monitoring.monitoring.models.DoesNotExist: Chart matching query does not exist.
```

<details>
  <summary>Trying to migrate master (description column not found). We can probably ignore this one by now but the migration 0013 in master in different from the one present in dev</summary>
```
./manage.py migrate
Operations to perform:
  Apply all migrations: account, admin, auth, check, config, connection, contenttypes, device_monitoring, geo, monitoring, notifications, openwisp_notifications, openwisp_users, pki, reversion, sessions, sites, socialaccount
Running migrations:
  Applying monitoring.0013_merge_traffic_metrics...Traceback (most recent call last):
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/sqlite3/base.py", line 396, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: no such column: monitoring_metric.description

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/core/management/base.py", line 369, in execute
[... 17 lines cut ...]
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/query.py", line 276, in __iter__
    self._fetch_all()
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/query.py", line 1261, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/query.py", line 57, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1151, in execute_sql
    cursor.execute(sql, params)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 100, in execute
    return super().execute(sql, params)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 68, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 77, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
  File "/home/pablo/.virtualenvs/openwisp-monitoring/lib/python3.6/site-packages/django/db/backends/sqlite3/base.py", line 396, in execute
    return Database.Cursor.execute(self, query, params)
django.db.utils.OperationalError: no such column: monitoring_metric.description
```
</details>

nepython, 2020-08-20:
@nemesisdesign, @PabloCastellano this issue can be closed right as the linked PR took care of the case? If I missing anything let me know, thanks.

nemesisdesign, 2020-08-20:
> @nemesisdesign, @PabloCastellano this issue can be closed, right? The linked PR took care of the case if I am correct. If I missing anything let me know, thanks.

Yes please close all the resolved issues which weren't closed automatically.

PR #154 (2020-07-07): [metric-config] Add fill metric configuration migration
Continuation to #91. There is a small change in `merge_traffic_migration` (I added it since, the new merged traffic metric will also have `field_name=rx_bytes`).

After the migration script,

```python
In [1]: from openwisp_monitoring.monitoring.models import Metric

In [2]: Metric.objects.first().configuration
Out[2]: 'ping'

In [3]: Metric.objects.first().related_fields
Out[3]: ['loss', 'rtt_min', 'rtt_max', 'rtt_avg']
```

So, it's fixed now with this for existing metrics :smiley:.

Closes #156
closed 2020-08-20 by nemesisdesign