openwisp/openwisp-monitoring #160: [dev] No metric configuration found for "used_memory"

nemesisdesign, 2020-07-08:
Happening on the dev branch.

I'm coming from master branch testing and when I switch to the dev branch and run migrations I get:

`No metric configuration found for "used_memory"`

Why does this happen?
Would this happen also for the other resource metrics?
Is it because some metrics have been renamed? Please make sure these cases are handled or I won't be able to upgrade existing systems.

nepython, 2020-07-08:
This can only happen if metric's `field_name` has been renamed from default ones as it's been used to map and fill configuration (refer https://github.com/openwisp/openwisp-monitoring/commit/6ad4bf2aacbee50476a5a742bdbe6d4c4114008d).

I didn't use `metric.name` for mapping since as you pointed it correctly, they can be renamed. I am unsure what I can use for mapping (in future we may use `configuration`) but for now I need some field.

nemesisdesign, 2020-07-27:
Closing for now, will reopen if needed.

closed 2020-07-27 by nemesisdesign