openwisp/openwisp-monitoring #167: [bug] This event loop is already running

nemesisdesign, 2020-07-20:
I'm getting this exception but only on production systems:

```
RuntimeError: This event loop is already running
  File "django/core/handlers/exception.py", line 34, in inner
    response = get_response(request)
  File "django/core/handlers/base.py", line 115, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "django/views/decorators/csrf.py", line 54, in wrapped_view
    return view_func(*args, **kwargs)
  File "django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)
  File "rest_framework/views.py", line 505, in dispatch
    response = self.handle_exception(exc)
  File "rest_framework/views.py", line 465, in handle_exception
    self.raise_uncaught_exception(exc)
  File "rest_framework/views.py", line 476, in raise_uncaught_exception
    raise exc
  File "rest_framework/views.py", line 502, in dispatch
    response = handler(request, *args, **kwargs)
  File "openwisp_monitoring/device/api/views.py", line 161, in post
    self._write(request, self.instance.pk)
  File "openwisp_monitoring/device/api/views.py", line 188, in _write
    self.instance.save_data()
  File "openwisp_monitoring/device/base/models.py", line 169, in save_data
    self.add_mac_vendor_info()
  File "openwisp_monitoring/device/base/models.py", line 150, in add_mac_vendor_info
    neighbor['vendor'] = self._mac_lookup(neighbor.get('mac_address'))
  File "cache_memoize/__init__.py", line 118, in inner
    result = func(*args, **kwargs)
  File "openwisp_monitoring/device/base/models.py", line 159, in _mac_lookup
    return mac_lookup.lookup(value)
  File "site-packages/mac_vendor_lookup.py", line 110, in lookup
    return self.loop.run_until_complete(self.async_lookup.lookup(mac))
  File "asyncio/base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "asyncio/base_events.py", line 425, in run_forever
    raise RuntimeError('This event loop is already running')
```

It looks like the mac vendor detection feature is done asynchronously and this is not playing well with the synchronous view in django.

nepython, 2020-09-23:
@nemesisdesign, can you please check if https://stackoverflow.com/a/56434301/13114765 solves this issue. What the code snippet does is not explained in the answer but in a comment below.

nemesisdesign, 2020-10-02:
I think the problem is being caused by the vendor makup library which is asynchronous.
I don't understand why something like this should be asynchronous.

There's another library which does this, it's called `netaddr`, at the time this feature was introduced the library seemed unmaintained but now it's being maintained again, so we could just switch to it.

So we should update these parts:

https://github.com/openwisp/openwisp-monitoring/blob/242c79370655092525b4df5429e058babdcc3703/requirements.txt#L9

https://github.com/openwisp/openwisp-monitoring/blob/242c79370655092525b4df5429e058babdcc3703/openwisp_monitoring/device/base/models.py#L172

Here's how to get the vendor with `netaddr`:
```python
from netaddr import EUI
EUI(value).oui.registration().org
```

@nepython can you help out with this?

PR #241 (2020-10-06): [fix] Switched mac vendor lookup library #167
Fixes #167

Checks:

- [x] I have manually tested the proposed changes
- [x] I have updated test cases
- [ ] I have updated the documentation (e.g. README.rst)
commit 54b3b58a (2020-10-06): [fix] Switched mac vendor lookup library #167

Fixes #167
closed 2020-10-06 by nemesisdesign