openwisp/openwisp-monitoring #169: [migrations] Squash migrations

nemesisdesign, 2020-07-25:
Once we're ready for the first release and once all the existing deployed systems using the development version have been updated, we should squash the migrations to make our life easier.

nepython, 2020-11-16:
@nemesisdesign, do you have any plans regarding upto what extent we should squash the migrations. Right now, I have squashed all 23 migrations in monitoring app and am testing it locally to see if there are any issues. Imo we can keep the migrations in device_monitoring app unsquashed and squash the [first two in check app](https://github.com/openwisp/openwisp-monitoring/tree/master/openwisp_monitoring/check/migrations) to ensure the ones having RunPython stay separate :)

atb00ker, 2020-11-17:
@nepython 
Ideally, we want to squash all the unreleased ones into one migrations file only.
However, this not being an ideal world, there are 2 cases where I can definitely think we need multiple migration files:
1. People are already using the module and squashing would mess up their existing systems.
2. Migration A is dependent  on Migration B.

Other than that there is no hard and fast rule of squashing migration or not.
If the migration file is getting too big, I suppose moving python functions separately would make sense.

Why don't you squash in the manner you think is the best and then let's look at the code to judge if it will be better to make any changes to it.

What do you think?

nemesisdesign, 2020-11-18:
Thanks for taking the time to ask @nepython.

I think we have to keep in mind these points:

- there are different instances using this unreleased version so we just have to make sure they'll be able to upgrade their systems
- some data migrations are useful for adding openwisp-monitoring to existing systems, eg:  https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/check/migrations/0005_create_config_applied.py, we can keep these as they are
- other data migrations are used just to convert the data from old DB schema to the new one, so we should be able to skip those
- we can keep unsquashed migration files in the repository and schedule them for removal in the 0.2 version

nepython, 2020-11-18:
Thanks @atb00ker @nemesisdesign, I went ahead to create #253 after taking in your feedback :)

PR #253 (2020-12-02): [migrations] Squash migrations #169
Closes #169
* All 23 migrations in `monitoring` app squashed to 1
* Squashed the first two migrations in `check` app. I think we can also replace migration 0004 in the `check` app by modifying this squashed initial migration. I wasn't very sure about it so didn't go ahead with the same.
* Squashed the two migrations in `device_monitoring` app.

Checks:

- [x] I have manually tested the proposed changes
- [ ]  ~~I have written new test cases to avoid regressions (if necessary)~~
- [ ] ~~I have updated the documentation (e.g. README.rst)~~
commit 8c473f13 (2020-12-02): [chores] Squashed migrations of pre 0.1 version #169

Closes #169
closed 2020-12-02 by nemesisdesign