openwisp/openwisp-monitoring #172: [ux] Help user to find out what's wrong

nemesisdesign, 2020-07-27:
Now that we have multiple checks and multiple metrics that generate alerts and change the health status to "problem", whenever a user accesses the device page and sees "problem" as health status, the first question that will arise to them is:

**What's the problem?**

We have to improve the UI so it becomes immediately evident where the problem lies.

Here's a possible solution (it's just a suggestion, if you think about a better solution let's discuss):

- we can add a new field like `health_status` in `openwisp_monitoring.device.admin.DeviceAdmin`
- this new field would give information about where the problem lies, this information can be retrieved by querying metrics related to the device which have the `is_healthy` field set to `False`, so the field could display the list of metrics which have problems
- this field could be shown only when the monitoring status is not ok (to save unneeded DB queries)

nepython, 2020-08-14:
@nemesisdesign, definitely this would be helpful to a user. I like the idea and had a suggestion that rather than adding a new field similar to `health-status` in device admin. How about rather having something like a tooltip text field that would be displayed whenever we hover the mouse over the `health-status`. Say when the user hovers over `PROBLEM` he / she gets a *lists of all unhealthy metrics*. If the user hovers over `OK` we display something like *all metrics are healthy*.
I came across this sometime back and I think we can implement the same if you like the idea, https://github.com/django-crispy-forms/django-crispy-forms/issues/389#issuecomment-74060287

nemesisdesign, 2020-08-14:
> @nemesisdesign, definitely this would be helpful to a user. I like the idea and had a suggestion that rather than adding a new field similar to `health-status` in device admin. How about rather having something like a tooltip text field that would be displayed whenever we hover the mouse over the `health-status`. Say when the user hovers over `PROBLEM` he / she gets a _lists of all unhealthy metrics_. If the user hovers over `OK` we display something like _all metrics are healthy_.
> I came across this sometime back and I think we can implement the same if you like the idea, [django-crispy-forms/django-crispy-forms#389 (comment)](https://github.com/django-crispy-forms/django-crispy-forms/issues/389#issuecomment-74060287)

I don't love the idea of the tooltip, because there are high changes that users will not discover this.
We need to give information at glance.
Also, I'm not sure if you planned on adding more dependencies, but we should avoid it.

nepython, 2020-08-17:
> I don't love the idea of the tooltip, because there are high changes that users will not discover this.
We need to give information at glance.

:+1:
>Also, I'm not sure if you planned on adding more dependencies, but we should avoid it.

Ok, when I checked earlier it seemed to be simple [html, js, css] without any external dependency, anyways since we aren't going forward with it, no worries.

nepython, 2020-08-17:
@nemesisdesign, does this seem to be good enough to help the user find the reason for problem status quickly?
We simply display the failing metric's names as a list in a device admin column next to Health status.
[image]

nemesisdesign, 2020-08-17:
> @nemesisdesign, does this seem to be good enough to help the user find the reason for problem status quickly?
> We simply display the failing metric's names as a list in a device admin column next to Health status.
> [image]

not in the list please, it's overkill there, in the device details page.

It should be a list like:

- {metric.is_healthy} (using the boolean image yes/no.svg that django provides) {metric.name}

nepython, 2020-08-17:
[..]
> not in the list please, it's overkill there, in the device details page.
> 
> It should be a list like:
> 
>     * {metric.is_healthy} (using the boolean image yes/no.svg that django provides) {metric.name}

This is already being displayed in the `AlertSettings Inline` tab :thinking: .
[image]
Honestly to me it seems a bit redundant, since the same thing would be displayed on 2 different tabs on the same page. The user just needs to change one tab (he / she won't mind that much effort, I think) :sweat_smile:.

nemesisdesign, 2020-08-17:
> [..]
> 
> > not in the list please, it's overkill there, in the device details page.
> > It should be a list like:
> > ```
> > * {metric.is_healthy} (using the boolean image yes/no.svg that django provides) {metric.name}
> > ```
> 
> This is already being displayed in the `AlertSettings Inline` tab  .
> [image]
> Honestly to me it seems a bit redundant, since the same thing would be displayed on 2 different tabs on the same page. The user just needs to change one tab (he / she won't mind that much effort, I think) .

When the user follows an alert link, they will be redirected to the overview tab.
They may not know that they have to click on alert settings to see this info, plus it doesn't really make sense to put it there.

Therefore yes we need this change and we may remove the "is healthy" info from the alert settings tab later on (**but not now**).

nepython, 2020-08-18:
@nemesisdesign, does a tabular representation like below seem good? I know styling needs to be improved :sweat_smile:, will work on it if it looks good. This field won't be displayed nor any queries be made to the db for the same if `Health Status` is not `PROBLEM`.
[image]

nemesisdesign, 2020-08-18:
> @nemesisdesign, does a tabular representation like below seem good? I know styling needs to be improved , will work on it if it looks good. This field won't be displayed nor any queries be made to the db for the same if `Health Status` is not `PROBLEM`.
> [image]

Very good!

A list would be enough, would look basically the same but simpler, I think it's easy to understand also without the table headers.
That would also allow to have the words closer to the icons.

nemesisdesign, 2020-08-31:
Closed by #201

PR #201 (2020-08-21): [ux] Help user to find out what's wrong #172

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)

[image]

Closes #172
closed 2020-08-31 by nemesisdesign