openwisp/openwisp-monitoring #173: [ux] Threshold value for resource metrics is weird

nemesisdesign, 2020-07-27:
We show the hardware resource monitoring charts (memory, cpu, storage) in percentages from 0% to 100% .

But the threshold values are 0.95, 0.8, 0.9 which is inconsistent.

Can we make the threshold values consistent? Can we use 95, 80 and 90?

nepython, 2020-08-14:
> But the threshold values are 0.95, 0.8, 0.9 which is inconsistent.
>
> Can we make the threshold values consistent? Can we use 95, 80 and 90?

That was what was being done originally (https://github.com/openwisp/openwisp-monitoring/pull/76#discussion_r426193740) :sweat_smile: .

In fact in order to have threshold values `0.95, 0.8, 0.9`, we had to change Threshold value field from Integer to Float.
This should be a doable, we would need to add one more migration for existing alertsettings. Let's do this after #185 which should be near complete hopefully and it would be more easier after it.

nepython, 2020-08-17:
I don't think we can do much about the old data. I read it at multiple places that `timeseries` data is not meant to be updated and thus tsdb are not optimized for the same. One such thread, https://community.influxdata.com/t/update-influxdb-measurements/8707

nemesisdesign, 2020-08-17:
> I don't think we can do much about the old data. I read it at multiple places that `timeseries` data is not meant to be updated and thus tsdb are not optimized for the same. One such thread, https://community.influxdata.com/t/update-influxdb-measurements/8707

I feared we'd end up deleting the old data, but it's fine, let's do it soon rather than later.

commit db11e3cc (2020-08-23): [ux] Improve threshold values of resource metrics #173

Closes #173
commit 66bfaa54 (2020-08-28): [ux] Improve threshold values of resource metrics #173

Closes #173
closed 2020-08-31 by nemesisdesign