openwisp/openwisp-monitoring #174: [bug] AppRegistryNotReady in dev after deploy

nemesisdesign, 2020-07-29:
@nepython I tried deploying the `dev` branch on a VM and got this when trying to run migrations (or any other management command)
```
Error in sys.excepthook:
Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 193, in _add_installed_apps_translations
    app_configs = reversed(list(apps.get_app_configs()))
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/apps/registry.py", line 144, in get_app_configs
    self.check_apps_ready()
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/apps/registry.py", line 135, in check_apps_ready
    raise AppRegistryNotReady("Apps aren't loaded yet.")
django.core.exceptions.AppRegistryNotReady: Apps aren't loaded yet.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.8/site-packages/raven/utils/serializer/manager.py", line 76, in transform
    return repr(value)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/functional.py", line 91, in __repr__
    return repr(self.__cast())
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/functional.py", line 136, in __cast
    return self.__text_cast()
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/functional.py", line 124, in __text_cast
    return func(*self.__args, **self.__kw)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/__init__.py", line 92, in gettext
    return _trans.gettext(message)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 354, in gettext
    _default = _default or translation(settings.LANGUAGE_CODE)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 267, in translation
    _translations[language] = DjangoTranslation(language)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 154, in __init__
    self._add_installed_apps_translations()
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 195, in _add_installed_apps_translations
    raise AppRegistryNotReady(
django.core.exceptions.AppRegistryNotReady: The translation infrastructure cannot be initialized before the apps registry is ready. Check that you don't make non-lazy gettext calls at import time.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/openwisp2/env/lib/python3.8/site-packages/raven/utils/serializer/manager.py", line 94, in transform
    return serializer.transform(value, **kwargs)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/raven/utils/serializer/manager.py", line 78, in transform
    logger.exception(e)
  File "/usr/lib/python3.8/logging/__init__.py", line 1467, in exception
    Convenience method for logging an ERROR with exception information.
  File "/usr/lib/python3.8/logging/__init__.py", line 1469, in exception
    self.error(msg, *args, exc_info=exc_info, **kwargs)
  File "/usr/lib/python3.8/logging/__init__.py", line 1455, in error
    Log 'msg % args' with severity 'ERROR'.
  File "/usr/lib/python3.8/logging/__init__.py", line 1463, in error
    self._log(ERROR, msg, args, **kwargs)
  File "/usr/lib/python3.8/logging/__init__.py", line 1577, in _log
    self.handle(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 1587, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 1649, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.8/logging/__init__.py", line 950, in handle
    self.emit(record)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/log.py", line 120, in emit
    message = "%s\n\n%s" % (self.format(no_exc_record), reporter.get_traceback_text())
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/views/debug.py", line 341, in get_traceback_text
    return t.render(c)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/template/base.py", line 171, in render
    return self._render(context)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/template/base.py", line 163, in _render
    return self.nodelist.render(context)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/template/base.py", line 936, in render
    bit = node.render_annotated(context)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/template/base.py", line 903, in render_annotated
    return self.render(context)
[... 58 lines cut ...]
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/opt/openwisp2/env/lib/python3.8/site-packages/openwisp_monitoring/monitoring/__init__.py", line 1, in <module>
    from .charts import register_chart, unregister_chart  # noqa
  File "/opt/openwisp2/env/lib/python3.8/site-packages/openwisp_monitoring/monitoring/charts.py", line 77, in <module>
    'unit': f' {_("ms")}',
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/functional.py", line 124, in __text_cast
    return func(*self.__args, **self.__kw)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/__init__.py", line 92, in gettext
    return _trans.gettext(message)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 354, in gettext
    _default = _default or translation(settings.LANGUAGE_CODE)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 267, in translation
    _translations[language] = DjangoTranslation(language)
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 154, in __init__
    self._add_installed_apps_translations()
  File "/opt/openwisp2/env/lib/python3.8/site-packages/django/utils/translation/trans_real.py", line 195, in _add_installed_apps_translations
    raise AppRegistryNotReady(
django.core.exceptions.AppRegistryNotReady: The translation infrastructure cannot be initialized before the apps registry is ready. Check that you don't make non-lazy gettext calls at import time.
```

When removing [this line](https://github.com/openwisp/openwisp-monitoring/blob/dev/openwisp_monitoring/monitoring/__init__.py#L1) the application can start.

I have no idea why this bug is not happening during development.
Why is that file being imported?
If there's anything that needs to be registered, do it in the ready() method of the AppConfig.

nepython, 2020-07-29:
>I have no idea why this bug is not happening during development.

Yes, I can't recreate it in development but the bug makes sense.
>When removing this line the application can start.

Ok, this line is not doing anything but easing the import. I had added it after the review here https://github.com/openwisp/openwisp-monitoring/pull/117#discussion_r436075290. So, removing it would mean the below simple change
```python
from openwisp_monitoring.monitoring import register_chart_configuration   # currently
```
**to**
```python
from openwisp_monitoring.monitoring.charts import register_chart_configuration
```

nemesisdesign, 2020-07-29:
Closed by 5094797

PR #176 (2020-07-29): [fix] Avoid register_chart, unregister chart import while apps are being loaded

Checks:

- [ ] I have manually tested the proposed changes (**can't reproduce locally**)
- [ ] I have written new test cases to avoid regressions (if necessary)
- [x] I have updated the documentation (e.g. README.rst)

Closes #174
commit 50947972 (2020-07-29): [fix] Fixed AppRegistryNotReady in monitoring apps.py #174

Fixes #174
commit 821ebe7b (2020-08-07): [fix] Fixed AppRegistryNotReady in monitoring apps.py #174

Fixes #174
closed 2020-07-29 by nemesisdesign