openwisp/openwisp-monitoring #203: Prepare 0.1 release

nemesisdesign, 2020-08-20:
Prerequisites:

- #201 
- #202 
- #199 
- #182
- #169 
- #167
- first release of openwisp-notifications

After this list is completed, we can go ahead and issue the first release! :clap:

PabloCastellano, 2020-08-25:
@nemesisdesign Nice to see that the release date is so close now! I think you already know it, but GitHub milestones are useful to track the pending issues for a release

nemesisdesign, 2020-08-26:
@PabloCastellano thanks for the suggestion, I created the milestone in this repo: https://github.com/openwisp/openwisp-monitoring/milestone/1
For the issues in the other repos we'll rely on the link here I guess.

nemesisdesign, 2020-11-11:
Only 4 issues left for 0.1 release!

nepython, 2020-11-14:
> Only 4 issues left for 0.1 release!

This is so close for an initial release now, I will try to work upon lua-monitoring issue right now :smile:

nemesisdesign, 2021-01-31:
Finally :+1:

closed 2021-01-31 by nemesisdesign