openwisp/openwisp-monitoring #207: [bug] Devices which do not have management_ip set should remain UNKNOWN

nemesisdesign, 2020-08-20:
While migrating the instance at the University of Urbino with @cappe87 and @AlessiaVe, we noticed that the devices initially were in the `UNKNOWN` status (correct), but after a bit of time, some of them started to show as `OK`, even though they didn't have their `management_ip` set, because the openwisp version used was too old to have that feature configured.

In this case, I think the correct behavior should have been the status `UNKNOWN` being maintained. I thought we already fixed this issue, but for some reason it's still manifesting somehow.

nepython, 2020-09-23:
I am not fully sure how `management_ip` works. Can you please guide me a little so I can get started with fixing this. Also is it documented anywhere that what was the behavior of the old version and how different it is exactly in newer versions of openwisp.

nemesisdesign, 2020-09-24:
@nepython the management_ip is used for monitoring and controller to reach the device via a management tunnel, VPN or interface, if it's not specified in most cases openwisp-monitoring should not try to contact the device unless `OPENWISP_MONITORING_MANAGEMENT_IP_ONLY` is set to `False`: https://github.com/openwisp/openwisp-monitoring#openwisp-monitoring-management-ip-only

nemesisdesign, 2020-10-23:
Tested again on a new system and this bug didn't show up, I assume it's solved.

closed 2020-10-23 by nemesisdesign