openwisp/openwisp-monitoring #218: [bug] New AlertSettings tolerance does not respect metric configuration

nemesisdesign, 2020-08-26:
It seems that when new `AlertSettings` objects are created, the `custom_tolerance` field is always set to zero because that's the default value:

https://github.com/openwisp/openwisp-monitoring/blob/dev/openwisp_monitoring/monitoring/base/models.py#L499

If the default tolerance is changed using `OPENWISP_MONITORING_METRICS`, new `AlertSettings` won't respect it and will instead set the `custom_tolerance` field to zero.

I think the default attribute should be removed from the model, that should cause the default to be `NULL`.

Please create a failing test first, the test could simply be to initialize an `AlertSettings` object and ensure its `custom_tolerance` value is None.

nepython, 2020-08-26:
@nemesisdesign, should I create a new migration or update the old one. Unsure since it is in `dev` branch.

nemesisdesign, 2020-08-26:
> @nemesisdesign, should I create a new migration or update the old one. Unsure since it is in `dev` branch.

The default value in the DB needs to be changed, and we can do that only with a migration.
Moreover, that default is present also in the master branch: https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/monitoring/base/models.py#L487-L489

PR #222 (2020-08-28): [fix] AlertSettings tolerance field should not have any default value

Checks:

- [ ] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)

Closes #218
commit 6db59d69 (2020-08-28): [fix] Removed default value from AlertSettings.custom_tolerance #218

Fixes #218
closed 2020-08-31 by nemesisdesign