openwisp/openwisp-monitoring #223: [change] Rethink when it's best to check threshold (eg: after metric is written)

nemesisdesign, 2020-08-29:
We recently changed the metric write operation to be run in the background #190, but the threshold is checked right after the write operation is sent to be performed to the background worker:

https://github.com/openwisp/openwisp-monitoring/blob/f6c75913ecadf8d17e4ba3d1a9c248cf11ff5689/openwisp_monitoring/monitoring/base/models.py#L207-L220

The `check_threshold` method handles a case in which past data is used for the check, but the code has not been updated after the changes to #190, so several inconsistencies may appear:

- the check threshold operation may be performed before the value is written to the tsdb, which can cause inconsistencies in behavior since the code has been written to be executed after the values have been written
- the check threshold may fail if in that moment the tsdb is experiencing issues

Taking this into account we should think about a solution.
I think we should find a way to ensure the check threshold operation is done after the metric has been written successfully to the tsdb.

nepython, 2020-09-23:
@nemesisdesign, can we do it this way.
1) Right after data is written to TSDB send a signal (say `post_timeseries_write`).
2) Add a new receiver listening to `post_timeseries_write` which will call `check_threshold` whenever the signal is emitted.

Few suggestions:-
* We can convert `check_threshold` method into a receiver but I think it (`check_threshold`) should retain its existence as a model  method since it is useful in some cases (eg. unit testing)
* One more thing that should we have a `post_timeseries_write` signal in the first place or simply reassign `post_write` signal this job. The reason being before we went for async write, this was I think precisely what `post_write` signal was showing (that is, a write operation has been completed in the tsdb). I am not very sure where exactly is post_write being used so can't visualize if this will break existing code, so please let me know accordingly.

nemesisdesign, 2020-09-24:
> @nemesisdesign, can we do it this way.
> 
> 1. Right after data is written to TSDB send a signal (say `post_timeseries_write`).
> 2. Add a new receiver listening to `post_timeseries_write` which will call `check_threshold` whenever the signal is emitted.

It should work! :+1: 

> Few suggestions:-
> 
> * We can convert `check_threshold` method into a receiver but I think it (`check_threshold`) should retain its existence as a model  method since it is useful in some cases (eg. unit testing)

I agree.

> * One more thing that should we have a `post_timeseries_write` signal in the first place or simply reassign `post_write` signal this job. The reason being before we went for async write, this was I [...]

Probably reassign it and update the docs.

nemesisdesign, 2020-10-06:
@nepython I have been hitting this issue while testing #240, it seems there's a weird interaction between the failure to write the results and the tolerance implementation.

nepython, 2020-10-06:
> @nepython I have been hitting this issue while testing #240, it seems there's a weird interaction between the failure to write the results and the tolerance implementation.

Cool, I work on this issue:smile:
@nemesisdesign, can you share how did you test #240 and what was the bug that was being encountered by you. I am asking to ensure we include good acid tests for the same :)

nemesisdesign, 2020-10-06:
> > @nepython I have been hitting this issue while testing #240, it seems there's a weird interaction between the failure to write the results and the tolerance implementation.
> 
> Cool, I work on this issue
> @nemesisdesign, can you share how did you test #240 and what was the bug that was being encountered by you. I am asking to ensure we include good acid tests for the same :)

@nepython: sure this is what I was doing (manual testing):

- set tolerance to 5 minutes
- make sure device is not reachable anymore (unplug cable or change management_ip to a bogus address)
- run checks, ensure no alert is generated
- set the address to a reachable address again
- run checks, ensure no alert is generated
- set again the address to a bogus one, then leave it and wait for the tolerated time to pass
- after 5 minutes I did not receive any alarm and I started monitoring the celery worker to see what was going on
- then I noticed the worker was not writing results straightaway but was retrying them, the bar chart was being correctly drawn as "unreachable" but the status was not changing, nor the alert was being generated
- I insisted several times running `./manage.py run_checks` until the metric was finally written without the need of a retry: at that point the status changed and the alert was generated

Very worrying.. I'm not sure what it's going on but I suspect it has to do with this issue, or an interaction of the two cases.

I'm not sure if ot may be possible to replicate this case in the tests by using a variation of this new test I introduced in #240:

https://github.com/openwisp/openwisp-monitoring/blob/54b3b58ad637370ab85346503d3c6752edee8a37/openwisp_monitoring/monitoring/tests/test_monitoring_notifications.py#L288-L316

I leave this link here in case it may be useful for you to test.

commit ee9b8033 (2020-10-24): [monitoring] Rethink check_threshold to be called after data is written to timeseries #223

Closes #223
commit f2f78d86 (2020-10-24): [monitoring] Rethink check_threshold to be called after data is written to timeseries #223

Closes #223
commit 2c4d3105 (2020-10-24): [monitoring] Rethink check_threshold to be called after data is written to timeseries #223

Closes #223
commit 6d7e41d8 (2020-11-04): [monitoring] Rethink check_threshold to be called after data is written to timeseries #223

Closes #223
closed 2020-11-11 by nemesisdesign