openwisp/openwisp-monitoring #233: is 'openwisp_monitoring.notifications' needed in INSTALLED_APPS?

pniaps, 2020-09-16:
I see it's used in https://github.com/openwisp/openwisp-monitoring/blob/bcc3ccd7f3a2ab892dac2e8cc1d3280b78026a93/tests/openwisp2/settings.py#L53 but is not documented in https://github.com/openwisp/openwisp-monitoring/blob/master/README.rst#setup-integrate-in-an-existing-django-project

What happends if not used? I see it has migrations, but don't know if they are executed/needed or not.

nepython, 2020-09-16:
@pniaps, notifications module was earlier included in the monitoring module. It has now been decoupled and is going to be extensively used along with other OpenWISP modules besides the monitoring module.
The migration scripts that are there in openwisp_monitoring.notifications were kept to have successful data migration from the app openwisp_monitoring.notifications to openwisp_notifications in some production instances which were already using openwisp_monitoring.notifications.

So yes, openwisp_monitoring.notifications is longer needed in INSTALLED_APPS.

I think we can remove it when monitoring module nears first release provided all production instances have successfully made a data migration. Wdyt @nemesisdesign @pandafy ?

pniaps, 2020-09-16:
Thanks for the explanation.

This documentation is for a new project, so it's correct. Maybe not in the test project (should be removed?).

nemesisdesign, 2020-09-17:
I think we should be able to remove it now, what do you think @pandafy?

pandafy, 2020-10-18:
Yeah, it should be completely fine to remove `openwisp_monitoring.notifications`

> I think we can remove it when monitoring module nears first release provided all production instances have successfully made a data migration.

I agree with @nepython on this, maybe we can remove it from installation documentations but keep the code and add a pointer in FAQs/Troubleshooting guide.

nemesisdesign, 2020-11-11:
I confirm It's safe to remove it!

nepython, 2020-12-17:
Why haven't we removed the migration files of the notifications app before closing this issue?
> I confirm It's safe to remove it!

If all production instances have been successfully upgraded to use `openwisp_notifications` then migration files can be deleted before the initial release imo.

nemesisdesign, 2020-12-17:
> Why haven't we removed the migration files of the notifications app before closing this issue?
> 
> > I confirm It's safe to remove it!
> 
> If all production instances have been successfully upgraded to use `openwisp_notifications` then migration files can be deleted before the initial release imo.

I didn't notice the migration files were not removed, I'm reopening!

PR #250 (2020-12-15): [fix] Removed 'openwisp_monitoring.notifications'
Removed 'openwisp_monitoring.notifications'
Upgraded openwisp_utils latest version
Renamed `mac_address` to `mac` and `ip_address` to `ip` for improving consistency
Closes #233

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
commit 5c85e805 (2020-12-15): [change] Removed 'openwisp_monitoring.notifications #233

Closes #233
PR #256 (2020-12-18): [refactor] Removed 'openwisp_monitoring.notifications from installed apps
Closes #233

Checks:

- [ ] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 8471af1d (2020-12-18): [chores] Removed openwisp_monitoring.notifications migrations #233

Closes #233
closed 2020-12-18 by nemesisdesign