openwisp/openwisp-monitoring #234: [monitoring] Support for tolerance in both directions

nemesisdesign, 2020-09-18:
Right now, in the following situation this is what happens:

- set tolerance of a device to 15 minutes
- the device goes down, after 15 minutes an alert is emitted
- the device goes up: a notification is immediately generated - **but it would be better to respect the tolerance here as well**

I suggest to proceed with writing the tests first.
Look at the existing tests:

- https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/monitoring/tests/test_monitoring_notifications.py#L194-L208
- probably this test will need to be modified: https://github.com/openwisp/openwisp-monitoring/blob/master/openwisp_monitoring/monitoring/tests/test_monitoring_notifications.py#L243-L256 - if I remember correctly it expects the alert to be generated straightaway, but once we implement this change the alert should not be generated unless the number of minutes specified in tolerance has passed

PabloCastellano, 2020-09-29:
I'm having some problems with this issue. So let's break it down and summarize how it should work. As example, let's think about an alert that the user configures to be notified then the metric "cpu load" goes above 90%.

The trivial case is when tolerance is 0. In this case, as soon as the threshold is crossed, either up (there's a problem) or down (the problem is now solved), an alert is sent to the user. I guess we agree on it, but let me know otherwise.

The other case is handling tolerance > 0 (minutes). The main goal of using this parameter is to avoid getting flooded by flapping alerts all the time. In this case, we want to check the metric values saved previously in the timeseries database to decide if we must alert the user or not. Considering 10 minutes tolerance and checks run every 5 minutes, there are different subcases for this scenario:

**The metric is healthy but the last check says that CPU load is above 90%**

Ok, so now we need to check the previous measurements to decide if we alert the user or not. We retrieve the measurements for the last 10 minutes (tolerance), which gives us two values and the following scenarios:

1. At t-5 and t-10, both values were >90 too. The user must be notified and `is_healthy=False` set.
2. At t-5 and t-10, both values were <90. In this case it is not enough to notify the user.
3. At t-5 the value is >90 and at t-10 the value is <90. Same case as before, but the user is closer to be notified. One more value over 90 and we are in case 1.
4. At t-5 the value is <90 and at t-10 the value is >90. The alert is clearly flapping. **Should we alert the user at this point or should we wait for 3 consecutive values over 90?**

I'm somewhat confused here about the is_healthy property. As I understand, we change it at the same time we send the alert, am I right? That is, it is not possible to have an unhealthy check without having been notified.

**The metric is unhealthy and the last check says that CPU load is above 90%**

Since it is unhealthy, the user has been already notified. We retrieve the measurements for the last 10 minutes (tolerance), which gives us two values and the following scenarios:

1. At t-5 and t-10, both values were >90 too. The user was notified and the situation has not changed. Do nothing.
2. At t-5 and t-10, both values were <90. Looks like the alert is flapping again. There hasn't passed enough time to notify the user. Do nothing.
3. At t-5 the value is >90 and at t-10 the value is <90. The alert is flapping, do nothing.
4. At t-5 the value is <90 and at t-10 the value is >90. The alert is flapping, do nothing.

Similarly, there are two more cases:

**The metric is healthy and the last check says that CPU load is below 90%**

1. Everything is under control. We don't even need to retrieve previous measurements. Do nothing.

**The metric is unhealthy and the last check says that CPU load is below 90%**

We retrieve the values for the previous 10 minutes, and:

1. At t-5 and t-10, both values were >90 too. The system is recovering but it is too early to notify the user. Do nothing.
2. At t-5 and t-10, both values were <90. The user must be notified (problem is resolved) and `is_healthy=True` set.
3. At t-5 the value is >90 and at t-10 the value is <90. The alert is flapping, do nothing.
4. At t-5 the value is <90 and at t-10 the value is >90. The alert is flapping, do nothing.

nemesisdesign, 2020-10-02:
> I'm having some problems with this issue. So let's break it down and summarize how it should work. As example, let's think about an alert that the user configures to be notified then the metric "cpu load" goes above 90%. Please let me know if I got it correctly.
> 
> The trivial case is when tolerance is 0. In this case, as soon as the threshold is crossed, either up (there's a problem) or down (the problem is now solved), an alert is sent to the user. I guess we agree on it, but let me know otherwise.
> 
> The other case is handling tolerance > 0 (minutes). The main goal of using this parameter is to avoid getting flooded by flapping alerts all the time. In this case, we want to check the metric values saved previously in the timeseries database to decide if we must alert the user or not. Considering 10 minutes tolerance and checks run every 5 minutes, there are different subcases for this scenario:
> 
> **1. The metric is healthy but the last check says that CPU load is above 90%**
> 
> Ok, so now we need to check the previous measurements to decide if we alert the user or not. We retrieve the measurements for the last 10 minutes (tolerance), which gives us two values and the following scenarios:
> 
> * 1.1. At t-5 and t-10, both values were >90 too. The user must be notified and `is_healthy=False` set.
> * 1.2. At t-5 and t-10, both values were <90. In this case it is not enough to notify the user.
> * 1.3. At t-5 the value is >90 and at t-10 the value is <90. Same case as before, but the user is closer to be notified. One more value over 90 and we are in case 1.
> * 1.4. At t-5 the value is <90 and at t-10 the value is >90. The alert is clearly flapping. **Should we alert the user at this point or should we wait for 3 consecutive values over 90?** question

No alert unless metric has been unhealthy for more than the time specified in tolerance. 

> I'm somewhat confused here about the is_healthy property. As I understand, we change it at the same time we send the alert, am I right? That is, it is not possible to have an unhealthy check without having been notified. 

It is used to know if a metric is healthy given the threshold settings.
It's used change the health status of the device, as well as to show in the device page the metrics which are having issues when the health status is not OK.
It's changed to False when the metric value crosses the threshold for the specified tolerance time or to True when it returns withing the threshold.
 
> **2. The metric is unhealthy and the last check says that CPU load is above 90%**
> 
> Since it is unhealthy, the user has been already notified. We retrieve the measurements for the last 10 minutes (tolerance), which gives us the following scenarios (**do nothing for all of them**):
> 
> * 2.1. At t-5 and t-10, both values were >90 too. The user was notified and the situation has not changed. Do nothing.
> * 2.2. At t-5 and t-10, both values were <90. Looks like the alert is flapping again. There hasn't passed enough time to notify the user. Do nothing.
> * 2.3. At t-5 the value is >90 and at t-10 the value is <90. The alert is flapping, do nothing.
> * 2.4. At t-5 the value is <90 and at t-10 the value is >90. The alert is flapping, do nothing.

I think this should be like 3: if the last measurements is unhealthy and the metric is unhealthy, the situation has not changed and we should do nothing.

> **3. The metric is healthy and the last check says that CPU load is below 90%**
> 
> * 3.1. Everything is under control. We don't even need to retrieve previous measurements. Do nothing.
> 
> **4. The metric is unhealthy and the last check says that CPU load is below 90%**
> 
> We retrieve the values for the previous 10 minutes, and:
> 
> * 4.1. At t-5 and t-10, both values were >90 too. The system is recovering but it is too early to notify the user. Do nothing.
> * 4.2. At t-5 and t-10, both values were <90. The user must be notified (problem is resolved) and `is_healthy=True` set.
> * 4.3. At t-5 the value is >90 and at t-10 the value is <90. The alert is flapping, do nothing.
> * 4.4. At t-5 the value is <90 and at t-10 the value is >90. The alert is flapping, do nothing.

commit 98eeae65 (2020-09-24): [fix] Support for tolerance in both directions #234

Fixes #234
commit 8bb117ac (2020-09-24): [fix] Support for tolerance in both directions #234

Fixes #234
commit 171ec74a (2020-10-03): [change] Support tolerance in both directions #234

- if a metric is flapping during the tolerated interval,
  the system will not trigger status changes
- if a metric is not flapping, the system will trigger
  a status change, whether recovery or not

Before this change, recoveries did not respect the tolerance,
causing the generation of many alerts on flapping metrics.

Closes #234
commit 5f333729 (2020-10-03): [change] Support tolerance in both directions #234

- if a metric is flapping during the tolerated interval,
  the system will not trigger status changes
- if a metric is not flapping, the system will trigger
  a status change, whether recovery or not

Before this change, recoveries did not respect the tolerance,
causing the generation of many alerts on flapping metrics.

Closes #234
closed 2020-10-06 by nemesisdesign