openwisp/openwisp-monitoring #239: Errors while running tests

devkapilbansal, 2020-10-01:
Running quality assurance tests, (installed from the development version of openwisp-utils), raises this error

[image]

devkapilbansal, 2020-10-01:
@nemesisdesign @atb00ker is this issue relevant? I was testing RST check for which I have to use the latest version of openwisp-utils.

nemesisdesign, 2020-10-02:
@devkapilbansal yes, looks like there's a RST syntax issue in that line. PS: next time use text instead of image, is more practical in these cases. Use images when showing UI issues.

devkapilbansal, 2020-10-02:
> @devkapilbansal yes, looks like there's a RST syntax issue in that line. PS: next time use text instead of image, is more practical in these cases. Use images when showing UI issues.

Updated. :+1:  Also, with RST syntax issues, there is one more right after that

vipermark7, 2020-10-16:
I'll take a crack at this! :)

commit 8b3ea6f1 (2020-11-05): [docs] Fixed README RST syntax warnings #239

Closes #239
closed 2020-11-05 by nemesisdesign