openwisp/openwisp-monitoring #265: [CI] Fix build

nemesisdesign, 2021-01-25:
Fix the build on github actions, which is failing.

nepython, 2021-01-25:
The build is failing because **influxdb** container might be taking longer than 15s to start on the CI.
@nemesisdesign, I think initially we had similar issue which we dealt with in #23.
I tried out the below steps locally to confirm if the changes introduced in #23 are still effective.
```bash
$ sudo service influxdb stop
$ ./tests/manage.py makemigrations --dry-run monitoring
Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 1 out of 5).
Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 2 out of 5).
Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 3 out of 5).
Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 4 out of 5).
Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 5 out of 5).
Traceback (most recent call last):
# and the urllib3 exception follows
```

nemesisdesign, 2021-01-25:
> The build is failing because **influxdb** container might be taking longer than 15s to start on the CI.
> @nemesisdesign, I think initially we had similar issue which we dealt with in #23.
> I tried out the below steps locally to confirm if the changes introduced in #23 are still effective.
> 
> ```shell
> $ sudo service influxdb stop
> $ ./tests/manage.py makemigrations --dry-run monitoring
> Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 1 out of 5).
> Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 2 out of 5).
> Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 3 out of 5).
> Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 4 out of 5).
> Got error while connecting to timeseries database. Retrying again in 3 seconds (attempt n. 5 out of 5).
> Traceback (most recent call last):
> # and the urllib3 exception follows
> ```

I don't see this output in the build output: https://github.com/openwisp/openwisp-monitoring/runs/1738993241

@devkapilbansal can you look into it when possible?
Also, I didn't notice this, but why is the redis service declared also in the deploy step, is it needed there?

PR #267 (2021-01-28): [ci] Fix CI build #265
## Description
- Added extra name option for influxdb image
- Change image to influxdb:alpine
- Remove unused redis

Fixes #265

Checks:

- [ ] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 11506b2a (2021-01-28): [ci] Fixed CI build #265

- Added extra name option for influxdb image
- Change image to influxdb:alpine
- Remove unused redis

Fixes #265
closed 2021-01-28 by nemesisdesign