openwisp/openwisp-monitoring #266: [cleanup] Remove custom swappable load_model

nemesisdesign, 2021-01-27:
Now that https://github.com/wq/django-swappable-models/pull/23 has been merged, the solution introduced in https://github.com/openwisp/openwisp-monitoring/pull/193 is not necessary anymore and can be removed.

sakshampathak1508, 2021-02-19:
Hi i am a python django developer  and would like to work on this issue i just want to make sure that in this issue the function load_model needs to be removed if that's the case i am ready to go. if not i could actually use some guidance

b-thebest, 2021-03-02:
Can I work on this issue ?

nemesisdesign, 2021-03-02:
I still can't find time to issue a new release of django-swappable-models, so this issue is on hold until then.

totallynotvaishnav, 2022-02-16:
I'm working on this.

nemesisdesign, 2022-02-16:
> I'm working on this.

Go for it, swapper has been released a few months ago.

PR #347 (2022-03-14): [chores] Removed custom swappable load_model #266
closes #266

Checks:

- [x] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 04d64477 (2022-03-31): [chores] Removed custom swappable load_model #266

Closes #266