openwisp/openwisp-monitoring #270: [feature] Chart signal strength (LTE/UMTS/etc.)

nemesisdesign, 2021-02-26:
Modem-manager interfaces are used for 3G/LTE routers and provide information regarding the protocols used in the mobile industry.

[OpenWISP Monitoring is already are able to collect information in the device status data](https://github.com/openwisp/openwisp-monitoring/commit/fb50d3dabbaf12fdc5b47f89a50507c2c0d25554) (please take an indepth look at this link), which is then displayed in the device status tab.

Now we need to add logic to the device metric view which is also able to log the signal strength (RSSI) information in a chart, as we already do for
the device charts.

The only difficult thing is that the RSSI value depends on the mobile protocol being used, because the router could be using
LTE, UMTS or other protocols depending on different factors.

I will clarify soon if we need to store only one value regardless of the protocol or if we should chart the RSSI for the different protocols.

Purhan, 2021-05-03:
I'm working on access technology chart but I'm not sure if my approach is good enough. I was thinking of using this query:
```python
    'access_tech': {
        'influxdb': (
            "SELECT MODE(access_tech) FROM {key} WHERE "
            "time >= '{time}' AND content_type = '{content_type}' AND "
            "object_id = '{object_id}' GROUP BY time(1d)"
        )
    },
```
Here, `MODE` will give us the most frequent access tech, but I'm not sure how practical this is.

nemesisdesign, 2021-05-03:
> I'm working on the access technology chart but I'm not sure if my approach is good enough. I was thinking of using this query:
> 
> ```python
>     'access_tech': {
>         'influxdb': (
>             "SELECT MODE(access_tech) FROM {key} WHERE "
>             "time >= '{time}' AND content_type = '{content_type}' AND "
>             "object_id = '{object_id}' GROUP BY time(1d)"
>         )
>     },
> ```
> 
> Here, `MODE` will give us the most frequent access technology in a given time period, but I'm not sure how practical this is. What if there are 2 types of access methods being used simultaneously, which one do we consider in that case?

The device can work with only 1 access technology at time, but if there were switches from one to the other during the time period being grouped, I think we can show the one which was used most.

PR #294 (2021-06-30): [feature] Chart mobile (LTE/5G/UMTS/GSM) signal strength #270

Checks:

- [x] I have manually tested the proposed changes
- [x] I have written new test cases to avoid regressions (if necessary)
- [x] I have updated the documentation (e.g. README.rst)

Closes #270
commit ffe99657 (2021-06-30): [feature] Added mobile (5G/LTE/UMTS/GSM) signal strength charts #270

Closes #270
closed 2021-06-30 by nemesisdesign