openwisp/openwisp-monitoring #274: [CI] Migrate CI build to influxdb 2.0

nemesisdesign, 2021-03-10:
I had to pin the inflxudb docker image used for testing to version 1.8.4 because the new version 2.0 breaks the tests:
https://github.com/openwisp/openwisp-monitoring/pull/273

However, this is not great: we want to test with the latest influxdb version and therefore it would be great to upgrade our code so that the test environment is compatible with the last influxdb version. It should be just a matter of setting the authentication credentials correctly.

devkapilbansal, 2021-04-09:
Hi @nemesisdesign the current library used here supports `InfluxDB 1.x` only. 
Another library ([influxdb-client](https://github.com/influxdata/influxdb-client-python)) only support `InfluxDB 1.8+`.

Would it be good to drop support for InfluxDB 1.7 or earlier ? :thinking:

nemesisdesign, 2021-04-09:
> Hi @nemesisdesign the current library used here supports `InfluxDB 1.x` only.
> Another library ([influxdb-client](https://github.com/influxdata/influxdb-client-python)) only support `InfluxDB 1.8+`.
> 
> Would it be good to drop support for InfluxDB 1.7 or earlier ?

@devkapilbansal I guess we have no other viable option, influxdb 1.8 works well so if we can support from 1.8 onwards that's good enough, what worries me is the change of library, do you know if the syntax is similar?

devkapilbansal, 2021-04-09:
Yes, as both libraries are maintained by `InfluxData`, syntax is quite similar

AbhigyaShridhar, 2022-03-22:
@nemesisdesign I cannot reproduce this issue. I changed the version of influxdb image to 2.0-alpine in both docker-compose.yml and workflows/ci.yml. All the tests run perfectly.

devkapilbansal, 2022-03-22:
@AbhigyaShridhar how did you checked it? I just checked and issue is reproducible:
https://github.com/devkapilbansal/openwisp-monitoring/runs/5649166266?check_suite_focus=true#step:10:17

Had it not been reproducible, the version have been updated a long time ago

nemesisdesign, 2022-03-22:
> @nemesisdesign I cannot reproduce this issue. I changed the version of influxdb image to 2.0-alpine in both docker-compose.yml and workflows/ci.yml. All the tests run perfectly.

I wish it was so simple @AbhigyaShridhar, last time we checked, in order to use Influxdb 2.0, we would have to switch to a different python library, please read the first lines of https://github.com/influxdata/influxdb-python#influxdb-python for more info.

AbhigyaShridhar, 2022-03-23:
> I wish it was so simple @AbhigyaShridhar, last time we checked, in order to use Influxdb 2.0, we would have to switch to a different python library, please read the first lines of [...]

Oh, ok I'll look into that

nemesisdesign, 2022-10-28:
One announcement which is surely going to impact this work:
https://www.influxdata.com/blog/influxdb-engine/

Kevinjil, 2023-10-16:
In case anyone else wants to try it, there is an **ugly** way to get it up and running with InfluxDB 2 and the [1.x compatibility API](https://docs.influxdata.com/influxdb/cloud/reference/api/influxdb-1x/). I used an NGINX reverse-proxy that filters out the incompatible API requests coming from OpenWISP monitoring. Of course, this means that you need to set up the retention policy mapping to buckets yourself.

The following NGINX configuration filters requests and insets an InfluxDB v2 API token:
```nginx
server {
    listen       8086;
    listen  [::]:8086;
    server_name  influxdb;

    location / {
        proxy_pass https://0.0.0.0;
        proxy_ssl_verify on;
        proxy_ssl_name influxdb.example.com;
        proxy_ssl_server_name on;
        proxy_set_header influxdb.example.com;
        proxy_set_header Authorization "Token changeme";
        client_body_buffer_size 128K;

        if ($arg_q ~ "^CREATE\+DATABASE") {
                return 200 '{}';
        }
        if ($arg_q ~ "^CREATE\+RETENTION") {
                return 200 '{}';
        }
        if ($arg_q ~ "^ALTER\+RETENTION") {
                return 200 '{}';
        }
        if ($arg_q ~ "^SHOW\+RETENTION") {
                return 200 '{"results":[{"statement_id":0,"series":[{"columns":["name","duration","shardGroupDuration","replicaN","default"],"values":[["autogen", "0s", "0s", 1, true], ["short", "0s", "0s", 1, false]]}]}]}';
        }
    }
}
```

commit 2e3043c0 (2024-05-31): [monitoring] Adding influxDB 2.x version support #274

Fixes #274
commit 97a0694e (2024-05-31): [fix] Workflow and test updated #274

Fixes #274
commit e2b24469 (2024-05-31): [fix] Workflow and test updated #274

Fixes #274
commit 295446da (2024-05-31): [fix] Workflow and test updated #274

Fixes #274