openwisp/openwisp-monitoring #285: [docs] Add quickstart tutorial to README

nemesisdesign, 2021-03-30:
We need a quickstart that helps first time users to set up and understand the monitoring system.

Let's add more points to this issue!

- recommend setting openwisp-config `management_interface` option
- explain how to trigger monitoring checks without management interface (valid only if openwisp is in the same layer2 network)
- explain how default checks are created
- explain difference between active checks and passive metric collection

TODO: add missing details.

minhng99, 2021-04-01:
I'd like the quickstart to mentioned about ansible installed into a virtual environment in `/opt/openwisp2` which all the other `openwisp-*` modules located at and it doesn't need to be installed on the real OS, this got me really frustrated at first.
Also about the `supervisorctl` is for restarting the service.

chanduaki, 2021-04-01:
## Summary
we are using the Freeradius Sucessfully for the Network Authentication & Authorisation
we want to monitor our device , we had configured the Openwisp monitoring on the same system to Track the CPU, RAM stats etc
When we are cloned the repo and added the device , it shows device is adding but there is no Stats in the Charts

#### Steps executed

*  Cloned the repo from github (https://github.com/openwisp/openwisp-monitoring)
	> git clone https://github.com/openwisp/openwisp-monitoring.git
* Navigated to path and build &run the docker
  > cd openwisp-monitoring
  >  docker-compose up --build
 
* Accessing the openwispmonitoring at localhost:8000 
* Login into portal with username: admin , password: admin

In the Admin Panel
* Navigated to Home › Network Configuration › Devices › Add Device
Name: mac address of the device
Organization:default
Key: given the Ubuntu Machine unique key(cat etc/machine-id)
Mac Address: mac address of the device
Model: CPU Model
Operating system:Ubuntu 20.04
SOC / CPU: Intel
Notes: NA
* **Configuration**
   Configuration: #1
   * Backend: OPENWRT
   * Configuration status:modified (its by default, it is a readable field)
   * Templates: 
       * Click on Plus and added a Template with configuration
       ```package system

        config system 'system'
	        option hostname 'OpenWRT'
	        option timezone 'UTC'
	        option zonename 'UTC'
        
        package network
        
        config globals 'globals'
        
        config interface 'samo'
	        option auto '1'
	        option enabled '1'
	        option ifname 'samo'
	        option mtu '1500'
	        option proto 'none'
     ```

> Need clarification about the configuration variables (key, value) here
we had given as key ping and value 30
can you tell what are the values does we have to given in the key, value

 * Under templates , selected Monitoring (default) , sample template
 * Clicked on SaVE
 * Device is Added Sucessfully
 * Device Shows State as Unkown, not changing to any other
     
   
#### Queries
* Where could we miss the steps during the implementation

nemesisdesign, 2021-04-01:
@chanduaki have you set `management_interface` in `/etc/config/openwisp`?

What happens if you input these commands on the OpenWRT device?

```
which netjson-monitoring
netjson-monitoring
which openwisp-monitoring
openwisp-monitoring
```

chanduaki, 2021-04-02:
@nemesisdesign 
i havent set `management_interface` in `/etc/config/openwisp`

```
 /etc/config/openwisp
bash: /etc/config/openwisp: No such file or directory
```

Can you please tell how can we set in '/etc/config/openwisp'. It sgows that No such file or directory

nemesisdesign, 2021-04-02:
@chanduaki did you install [openwisp-config](https://github.com/openwisp/openwisp-config#openwisp-config) in your OpenWRT device?

commit ed731168 (2022-04-25): [change] Migrated to OpenWrt OpenWISP Monitoring package #285 #298 #306

- ensured we don't mess with updating openwisp config anymore
- renamed the scripts installed by the template so they begin with legacy-
- the script legacy-openwisp-monitoring should check whether openwisp-monitoring exists, and in that case it 
  shall just exit without doing anything
- made sure that on new installations, the monitoring template is not installed at all, hence we have to make the migrations smart so that they change the monitoring template only if installed, but the old migrations that install the monit [... cut]
closed 2022-04-25 by nemesisdesign