openwisp/openwisp-monitoring #290: [feature] Add REST API endpoints for device which include monitoring info

nemesisdesign, 2021-04-06:
Extend the [REST API of OpenWISP Controller](https://github.com/openwisp/openwisp-controller/pull/386#pullrequestreview-628538950)  by adding two endpoints, one for the list of devices and one for the device detail.

- `/api/v1/monitoring/device/`: list devices and include the monitoring status
- `/api/v1/monitoring/device/{pk}/`: show device details, this endpoint already exists but only shows chart data and optionally device status data (if status=true is passed in the querystring). We need to add the device and config data and allow accessing this info by using the user token (now only device key can access it), however only the authentication with the device key should be able to post monitoring data, because that's reserved for devices.

Saurav-Shrivastav, 2021-04-11:
For the endpoint `/api/v1/monitoring/device/{pk}/`, should the `GET` requests work without a device key and only rely on the user token or the user token should also be required along with the device key?

ManishShah120, 2021-04-11:
Hi @Saurav-Shrivastav for a `GET` request user token will do, but the endpoint should not allow `POST` request without the  authentication with the device key,

Manjuphoenix, 2021-11-21:
Hey @nemesisdesign is this issue still there?
I would like to work on this, can you assign me to this particular issue?

Manjuphoenix, 2021-11-21:
@nemesisdesign In the below picture i could see just 3 API's with predefined key values passed, if given more details regarding the different devices i can create API's for the same
[image]

nemesisdesign, 2021-11-24:
@Manjuphoenix I do not understand your questions, I think the issue description has enough details, if something is not clear, please ask a more specific question like @Saurav-Shrivastav did.

Anybody: you're free to work on this issue if you want, just let us know and send a PR.

Join https://gitter.im/openwisp/development if you need help.

uzaxirr, 2023-01-22:
Hey @nemesisdesign !! 
This is still open correct? I'll start working on it soon after the confirmation!

Aryamanz29, 2023-01-23:
Hey @uzaxirr! There is already a `draft PR` opened for this issue, #480, but I have put this work on hold because I have some queries regarding the API design that need to be clarified first before proceeding further. I would suggest you check `"good first issues"` on the [OpenWISP Contributor's Board](https://github.com/orgs/openwisp/projects/3.), thanks!

commit b56c3d25 (2023-03-02): [feature] Added REST API for device that includes monitoring info #290

Closes #290
commit 05750fba (2023-03-20): [feature] Added REST API for device that includes monitoring info #290

Closes #290
commit a39c4948 (2023-03-22): [feature] Added REST API for device list and detailed monitoring info #290

Closes #290
closed 2023-03-22 by nemesisdesign