openwisp/openwisp-monitoring #300: Migrate from Popen() to run()

nickberry17, 2021-05-25:
From the [Python docs](https://docs.python.org/3/library/subprocess.html#using-the-subprocess-module)
> The recommended approach to invoking subprocesses is to use the `run()` function for all use cases it can handle. For more advanced use cases, the underlying Popen interface can be used directly.
> 
> The run() function was added in Python 3.5; if you need to retain compatibility with older versions, see the Older high-level API section.

Is this an "advanced use case"?  Should we consider migrating to `run()`?

https://github.com/openwisp/openwisp-monitoring/blob/1b78b37b9e511db9029007111d1ddc2b97ba1a2b/openwisp_monitoring/check/classes/ping.py#L140

nemesisdesign, 2021-05-26:
@nickberry17 I'm not such in a hurry to change it, but should be just a 2 line change: https://github.com/openwisp/openwisp-monitoring/blob/1b78b37b9e511db9029007111d1ddc2b97ba1a2b/openwisp_monitoring/check/classes/ping.py#L140-L141
Initially this module supported also python2 but now we should be able to simplify that.

Aryamanz29, 2021-12-22:
I would like to work on this

PR #342 (2022-02-17): [refactoring] Migrate from Popen() to run() #300
Fixes #300

Checks:

- [x] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [ ] I have updated the documentation (e.g. README.rst)
commit 1ee01013 (2022-02-17): [refactor] Migrate from Popen() to run() #300

Fixes #300
closed 2022-02-17 by nemesisdesign