openwisp/openwisp-monitoring #301: [change] Reachable bar chart: show different color for barely reachable

nemesisdesign, 2021-05-27:
When reachability is higher than 0% but not yet to the right percentage level to show the yellow color, the color stays red, but this is problematic because at glance it looks like the device is still down while it may have just started to show as reachable.

Therefore we'll need to make sure the color shown is a bit different (dark orange?).

atb00ker, 2022-03-20:
How about a dark grey color for non-reachable device and red for low %.
Red for completely unreachable and blue for low %.
Or 
Red for completely unreachable and red with yellow strips for low %.

Having completely different colors or strips will ensure that someone with slight color blindness can also distinguish easily. 😄

yashikajotwani12, 2022-04-12:
Hey @nemesisdesign, I have one doubt that do we need to mention barely reachable like this :
[image]
Or 
We need  `dark orange color` only to show when the device is barely reachable and not to mention here.

nemesisdesign, 2022-04-12:
> Hey @nemesisdesign, I have one doubt that do we need to mention barely reachable like this : [image] Or We need `dark orange color` only to show when the device is barely reachable and not to mention here.

@yashikajotwani12: we can name it "Mostly unreachable" and it can be of dark orange color like that, including it in the legend.

We also need to add **"Mostly reachable"** if it's more than 90% but less than 99%. I would use the current reachable color for "Mostly reachable" and change the Reachable color to a darker green. I will improve the colors if needed in the end.

yashikajotwani12, 2022-04-12:
> @yashikajotwani12: we can name it "Mostly unreachable" and it can be of dark orange color like that, including it in the legend.
>
> We also need to add **"Mostly reachable"** if it's more than 90% but [...]

Thank you @nemesisdesign! I understood everything would work accordingly.

PR #355 (2022-04-29): [change] Reachable bar chart #301
Added Mostly Unreachable and
Mostly Reachable

Fixes #301
[image]

Checks:

- [ ] I have manually tested the proposed changes
- [ ] I have written new test cases to avoid regressions (if necessary)
- [X] I have updated the documentation (e.g. README.rst)
commit 5318baed (2022-04-29): [change] Added more shades to reachable chart #301

Added the following shades:

- Mostly Reachable
- Mostly Unreachable

Closes #301
closed 2022-04-29 by nemesisdesign