openwisp/openwisp-monitoring #303: [qa] Enable linting for css and js

niteshsinha17, 2021-06-10:
Linting for css and js is disabled by default. Check this https://github.com/openwisp/openwisp-utils/#openwisp-qa-check and enable it for ``monitoring`` and remove any previous code that is checking css and js code. Also, update readme.

Aryamanz29, 2021-09-29:
@niteshsinha17 I would like to work on this

niteshsinha17, 2021-09-29:
@Aryamanz29 great! Go ahead

Aryamanz29, 2021-10-02:
> @Aryamanz29 great! Go ahead

Hey @niteshsinha17  Do I need to add this `--csslinter --jslinter` [in this shell script at line 24th](https://github.com/openwisp/openwisp-monitoring/blob/8f8d47c2437fe2a89482b10951a20c55b81514aa/run-qa-checks#L23)?

Aryamanz29, 2021-10-02:
> > @Aryamanz29 great! Go ahead
> 
> Hey @niteshsinha17 Do I need to add this `--csslinter --jslinter` [in this shell script at line 24th](https://github.com/openwisp/openwisp-monitoring/blob/8f8d47c2437fe2a89482b10951a20c55b81514aa/run-qa-checks#L23)?

I think this change gonna enable linting , ref : https://github.com/openwisp/openwisp-utils/blob/9e182fd051e1b2142459014a9e5f6dc71a884ffa/openwisp-qa-check#L191  

`
--csslinter
    SKIP_CSS_LINTER=false
  --jslinter
    SKIP_JS_LINTER=false
`

devkapilbansal, 2021-10-03:
> I think this change gonna enable linting , ref : https://github.com/openwisp/openwisp-utils/blob/9e182fd051e1b2142459014a9e5f6dc71a884ffa/openwisp-qa-check#L191
>
> `--csslinter SKIP_CSS_LINTER=false [...]

@Aryamanz29 the comment you did above was correct. You need to add flags for csslinter and jslinter

PR #331 (2022-02-01): [qa] Enable css and js linting checks #303
Fixes #303
commit 6f884c9c (2022-02-01): [qa] Enable css and js linting checks #303

Fixes #303