openwisp/openwisp-monitoring #307: [change] If a device goes offline, we should clear the list of connected devices

nemesisdesign, 2021-06-22:
The device status gets cleared after 24 hours, so if a device goes offline the connected wifi clients will still be shown in the device status tab.

I am thinking that as soon as a device goes in critical conditions we could write a new device status with the wifi clients removed.

prafullpandey614, 2025-10-26:
Hi @nemesifier,
I'd love to contribute to this! It makes total sense to proactively clear the connected WiFi clients list on critical/offline status to avoid stale data in the UI. I can see a few ways to approach this:

- Hook into the critical condition detection (for ex- via the device's heartbeat or status update event) and trigger an immediate status update that omits the clients array.
- Ensure it's idempotent so it doesn't spam updates if the device flickers offline/online.

I can draft a PR with tests for the status update logic. Any specific code paths or edge cases I should watch for?
Thanks!

bhabishnu, 2026-10-27:
Hi @nemesifier, I'd like to work on this.

I checked the current path and it looks like when a device becomes critical, the WiFi sessions are closed but the last `DeviceData` snapshot/cache can still contain the associated clients.

Before I start on the fix: should this clear only the WiFi client data, or other volatile client data as well?